adafruit/circuitpython #9956: Metro ESP32-S3 with 16 MB Flash 8 MB PSRAM, no CIRCUITPY

dhalbert, 2025-01-13:
Reposted from comment in #9889:

>             On my Adafruit Metro ESP32-S3 with 16 MB Flash 8 MB PSRAM, all versions **after** adafruit-circuitpython-adafruit_metro_esp32s3-en_US-20241206-main-PR9873-c41d452.uf2 (**#9873**) fail to restart CIRCUITPY upon flashing the UF2 and performing a hard reset of the board. I know there was a discussion on Issue #9888, and the thought was that PR9889 had fixed the issue but not for the Metro ESP32-S3 board. Upon flashing the UF2, CIRCUITPY does reappear, but if you reset the board, the volume fails to mount. If you go back to CircuitPython 9.2.1, it will work great. Try to upgrade to CP 9.2.2, but you get this error where the board falls back into JTAG mode. I suspect the underlying bug is not yet fixed.

_Originally posted by @dbiersach in https://github.com/adafruit/circuitpython/issues/9889#issuecomment-2586002168_

dhalbert, 2025-01-13:
I am unable to duplicate this on my own Metro ESP32-S3. Could you try a clean install from here: https://circuitpython.org/board/adafruit_metro_esp32s3/ ? Copy off anything in CIRCUITPY you need to save. Then hold the BOOT button and press the RESET button. Let go of BOOT, and press the "OPEN INSTALLER" button on that page. Let it erase the flash. You will need to reset the board twice, once when it asks you to select the METROS3BOOT drive, and once when it asks you to select the CIRCUITPY drive.

If that still does not work, could you take a close-up picture of the silver module on the board so we can read the writing? Or transcribe it. When did you buy this board? Is there a B in a circle near pin 5? Thanks.

dbiersach, 2025-01-13:
Thanks. I wiped the Metro ESP32-S3 (after putting it into METROS3BOOT, which does appear) using metro-esp32-s3-factory-reset.UF2. I also tried applying metro-esp32-s3-factory-reset.bin using the Adafruit ESPTool and also tried applying combined.bin from tinyuf2-adafruit_metro_esp32s3-0.20.1.zip - all with the same effect. adafruit-circuitpython-adafruit_metro_esp32s3-en_US-9.2.1.uf2 works fine after a cold MCU boot (CIRCUITPY appears) while applying adafruit-circuitpython-adafruit_metro_esp32s3-en_US-9.2.2.uf2 results in no CIRCUITPY from appearing. I have no boot.py that could be calling storage.disable_usb_drive(). I've attached a photo of my chip. The board is ESP32-S3 N16R8 so I believe I have the latest stepping.
[image]

dbiersach, 2025-01-13:
I should add there *is* a small B by pin #5085

dbiersach, 2025-01-13:
I bought this from Adafruit on November 10, 2023 and attached is the output from ESPTool before flashing.
[image]

dhalbert, 2025-01-13:
Thanks for your detailed report. I have the same vintage Metro, with exactly the same ID numbers on the can. I also have the same esptool output. We saw a similar report of an issue with a Feather 4/2 as well. The user in that case had a newer rev board. There's a long thread starting here: https://discord.com/channels/327254708534116352/537365702651150357/1327358153477722165

What is your host computer, and what is the OS and version on it? If it's a PC, could you give the brand and model number and the CPU? It's possible this is some kind of host USB issue. The user above tried it on an RPi 3B running bullseye and a an older macOS machine.

I am using this on an Ubuntu 24.04 system running on a Dell Optiplex i7-8700.

dhalbert, 2025-01-13:
Aha! I have reproduced the problem when plugging directly into a USB2 or USB3 port. Previously I was using a hub.

dbiersach, 2025-01-13:
Via some bisection, I narrowed it down to a change between 2024-12-06T15:00:13.000Z adafruit-circuitpython-adafruit_feather_esp32s3_4mbflash_2mbpsram-en_US-20241206-main-PR9873-c41d452.uf2 (which works) and 2024-12-06T19:44:29.000Z adafruit-circuitpython-adafruit_feather_esp32s3_4mbflash_2mbpsram-en_US-20241206-main-PR9841-5c09b90.uf2 (which does not work)

dhalbert, 2025-01-13:
This will be a high priority for me to look at tomorrow. It's getting late here. Thanks!

dhalbert, 2025-01-13:
Metro ESP32-S2 9.2.2 is OK with or without hub.

fausterix, 2025-01-17:
Hi,
the same happens (no CIRCUITPY folder) after flashing 9.2.2 on a VCC-GND YD ESP32-S3-DevKitC-1 N16R8 (clone).
With 9.2.1 no problems.

Thank you

BoryaGames, 2025-01-17:
I have ESP32-S3-N16R8 and same problem:
- I have installed TinyUF2 first.
- Then I have installed CircuitPython 9.2.2.
- CIRCUITPY drive appears.
- After reset, there's no drive anymore and I can't connect to the serial too.
USB hub workaround works for me 👍

dhalbert, 2025-01-17:
#9971 may also be related, since STM also uses synopsys dwc2.

closed 2025-01-17 by dhalbert

pagong, 2025-01-18:
> I'm experiencing the same issue (no CIRCUITPY drive after reset) with CPy-9.2.2 on two Waveshare boards (WS ESP32-S3-Zero and WS ESP32-S3-Matrix) when plugged directly into a Linux laptop (Mint-22). However, [@dhalbert](https://github.com/dhalbert) 's workaround is working for me: plugging the boards into an intermediate USB-Hub (e.g. an HP-Dock) makes the CIRCUITPY drive appear

Problem is fixed with the new CPy-9.2.3 on both Waveshare boards: 
CIRCUITPY appears with direct USB connections (as well as with intermediate hub).