adafruit/circuitpython #9960: install support for ESP32-DevKitC-v4 ESP32-wroom-32u

stanelie, 2025-01-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.2
```

### Code/REPL

```python
N/A
```

### Behavior

Hello.
I need some help installing circuitpython for this ESP32_DevKitC_v4 board I got from Aliexpress. It looks exactly like [this one ](https://circuitpython.org/board/espressif_esp32_devkitc_v4_wrover/), but the ESP module is ESP32-wroom-32U . I was not able to flash it with the flash tool on that webpage. However, I was able to flash it with https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ using the bin file for the Huzzah32, which is also a wroom-32 with 4mb. It boots. However, I am worried the pins will be wrongly identified and stuff will not work. For example, my board does not have a neopixel, but the Huzzah firmware I flashed says it does. Is there a better alternative?

dhalbert, 2025-01-15:
Try this: https://circuitpython.org/board/espressif_esp32_devkitc_v4_wroom_32e/  . The title on the board description is wrong! It should be WROOM, not WROVER. I'll get that fixed.

GitHub issues are not meant for support like this. Better to use https://adafru.it/discord or https://forums.adafruit.com/viewforum.php?f=60

closed 2025-01-15 by dhalbert

stanelie, 2025-01-15:
It does not work.
It clic "continue" after choosing the port and connecting, but this next steps goes super fast, not like my other flash with other ESP32 boards, and it indeed does nothing, the firmware is not changed.

However, I was able to flash the https://circuitpython.org/board/espressif_esp32_devkitc_v4_wroom_32e/ file with the flash tool here https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ . 

My board now boots with `Adafruit CircuitPython 9.2.2 on 2025-01-09; Espressif ESP32 DevKitc V4 WROOM-32E with ESP32` , which is a lot closer to the 32U I truly have.

Thanks!!

dhalbert, 2025-01-15:
Glad it works. The E vs U is just the antenna connector.

stanelie, 2025-01-15:
Great! Thanks again!!