adafruit/circuitpython #9968: Blue led blinking.

0wwafa, 2025-01-17:
I installed Circuitpython 9.2.2 on my doit_esp32_devkit_v1 from here https://circuitpython.org/board/doit_esp32_devkit_v1/

After setting it up, everything seems to work. BUT the blue led blinks every 4 seconds.

How to disable it? And WHY is it blinking on itself?

tannewt, 2025-01-17:
Please ask questions like this on [the Discord](https://adafru.it/discord). This is likely the status LED. CircuitPython blinks every 5 seconds to show that no user code is running. 1 means code exited ok. 2 is an error. 3 is safe mode.

closed 2025-01-17 by tannewt

0wwafa, 2025-01-17:
> Please ask questions like this on [the Discord](https://adafru.it/discord). This is likely the status LED. CircuitPython blinks every 5 seconds to show that no user code is running. 1 means code exited ok. 2 is an error. 3 is safe mode.

I think it is DUMB. No code runnign should be doing nothing and going into low power mode.

dhalbert, 2025-01-17:
There's no reason to be rude. You can zero out the brightness of the status LED: https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime.rgb_status_brightness

0wwafa, 2025-01-17:
> There's no reason to be rude. You can zero out the brightness of the status LED: https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime.rgb_status_brightness

I didn't mean to sound rude. Sorry if it did.