adafruit/circuitpython #9971: CircuitPython 9.2.2 on WeAct STM32F411CE BlackPill (with flash) no CIRCUITPY folder and freeze the dev board.

fausterix, 2025-01-17:
Hi!

Flashing adafruit-circuitpython-stm32f411ce_blackpill_with_flash-en_US-**9.2.2**.bin on a WeAct STM32F411CE BlackPill 
(buyed on the official WeAct store on AliExpress) no CIRCUITPY folder appear and the MCU get stuck.

With adafruit-circuitpython-stm32f411ce_blackpill_with_flash-en_US-9.2.1.bin no problems, it works as expected.

On DFU-boot:
[mar dic 17 11:50:47 2024] usb 1-4: new full-speed USB device number 21 using xhci_hcd
[mar dic 17 11:50:47 2024] usb 1-4: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
[mar dic 17 11:50:47 2024] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[mar dic 17 11:50:47 2024] usb 1-4: Product: STM32  BOOTLOADER
[mar dic 17 11:50:47 2024] usb 1-4: Manufacturer: STMicroelectronics
[mar dic 17 11:50:47 2024] usb 1-4: SerialNumber: 315536863533

After flashing **9.2.1**:
[mar dic 17 11:52:35 2024] usb 1-4: USB disconnect, device number 21
[mar dic 17 11:52:41 2024] usb 1-4: new full-speed USB device number 22 using xhci_hcd
[mar dic 17 11:52:41 2024] usb 1-4: New USB device found, idVendor=239a, idProduct=006a, bcdDevice= 1.00
[mar dic 17 11:52:41 2024] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[mar dic 17 11:52:41 2024] usb 1-4: Product: stm32f411ce blackpill with flash
[mar dic 17 11:52:41 2024] usb 1-4: Manufacturer: WeAct
[mar dic 17 11:52:41 2024] usb 1-4: SerialNumber: 540022001551333532363331
[mar dic 17 11:52:41 2024] cdc_acm 1-4:1.0: ttyACM2: USB ACM device
[mar dic 17 11:52:41 2024] usb-storage 1-4:1.2: USB Mass Storage device detected
[mar dic 17 11:52:41 2024] scsi host6: usb-storage 1-4:1.2
[mar dic 17 11:52:42 2024] scsi 6:0:0:0: Direct-Access     WeAct    stm32f411ce blac 1.0  PQ: 0 ANSI: 2
[mar dic 17 11:52:42 2024] sd 6:0:0:0: Attached scsi generic sg3 type 0
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] 16377 512-byte logical blocks: (8.39 MB/8.00 MiB)
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] Write Protect is off
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] Mode Sense: 03 00 00 00
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] No Caching mode page found
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] Assuming drive cache: write through
[mar dic 17 11:52:42 2024]  sdc: sdc1
[mar dic 17 11:52:42 2024] sd 6:0:0:0: [sdc] Attached SCSI removable disk

Thank you

dhalbert, 2025-01-17:
I think this is another manifestation of #9956. STM chips also use the newly refactored dwc2 driver.
@hathach this could be another testbed.

hathach, 2025-01-20:
@dhalbert maybe bump up tinyusb could fix this, there is a couple of dwc2 fixed since last pull from cpy https://github.com/hathach/tinyusb/pull/2901 https://github.com/hathach/tinyusb/pull/2891

closed 2025-01-21 by dhalbert

fausterix, 2025-01-26:
Hi!

Actually even with CP 9.2.3 I'm having the same issue.

Thanks and regards

dhalbert, 2025-01-26:
@fausterix I added a fix for this post 9.2.3. Could you try the aboslute latest build from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/stm32f411ce_blackpill_with_flash/ ?

fausterix, 2025-01-27:
> [@fausterix](https://github.com/fausterix) I added a fix for this post 9.2.3. Could you try the aboslute latest build from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/stm32f411ce_blackpill_with_flash/ ?

Hi!
Just tested https://adafruit-circuit-python.s3.amazonaws.com/bin/stm32f411ce_blackpill_with_flash/en_US/adafruit-circuitpython-stm32f411ce_blackpill_with_flash-en_US-20250126-main-PR9996-52b5b2d.bin and **it works**!
Thank you!

PR #9985 (2025-01-21): update TinyUSB to incorporate latest dwc2 fixes
- Fixes #9971.

Updates TinyUSB to fix some `dwc2` issues: there was an off-by-one error in counting endpoints. This caused immediate problems on low-endpoint boards.

Tested on Feather STM32F405 and also re-tested on Metro ESP32-S3 to check for any regressions.

Thanks @hathach!