adafruit/circuitpython #9972: ESP32 microcontroller.cpu.temperature is None

0wwafa, 2025-01-17:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.2 on 2025-01-09; Ai Thinker ESP32-CAM with ESP32
```

### Code/REPL

```python
import microcontroller
microcontroller.cpu.temperature
```

### Behavior

Nothing happens. no output.

P.S.
It works in micropython.

The same happens with another ESP32 board I have.

In espruino it was removed and then re-added a couple of days ago:
https://github.com/espruino/Espruino/issues/2585

### Additional information

Albeit inaccurate, it's still a useful function.

dhalbert, 2025-01-17:
This functionality was removed from ESP-IDF some time ago due to not working well at all: https://github.com/espressif/esp-idf/issues/146. It was `temprature_sens_read()`. Yes, that was the actual spelling.

MicroPython returns the raw temperature reading by reading some registers. Due to the the inaccuracy and instability, I think we will skip that. It would be a support burden for us.

Re https://github.com/espruino/Espruino/issues/2585: it looks like it was put back for chips that support it (ESP32-C3, etc.).

closed 2025-01-17 by dhalbert

0wwafa, 2025-01-18:
The function works perfectly on WROOM-32 (devkit v1) and also on EPS32 CAM I have.

I see no burden nor a problem.
Just copy the code from micropython....  no matter if it's an internal temperature and to many it might seem "wrong". It's not on my devices. And they are pretty standard and very similar even if the esp32 cam form factor is different.

ESP32-CAM
```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```
Esp32-WROOM-32
```
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

both have 4MB of flash and they both work with micropython and circuitpython.

Please, @dhalbert re-enable the temperature.

dhalbert, 2025-01-18:
As mentioned here: https://github.com/espressif/esp-idf/issues/146#issuecomment-492570221,

> Unfortunately due to performance issues the temp sensor feature is no longer supported.

Espressif discontinued support of reading the ESP32 temperature  (for specifically that chip). We are not going to add it back because if it's unreliable it can give faulty and misleading values.

0wwafa, 2025-01-19:
> if it's unreliable it can give faulty and misleading values.

On both my devices it's not unreliable at all!