adafruit/circuitpython #9975: Setting cpu frequency below 80 Mhz messes up the serial console.

0wwafa, 2025-01-18:
At first I thought it crashed.. instead it's just the serial console messed up.

Adafruit CircuitPython 9.2.2-14-gab5c17d864 on 2025-01-16; Ai Thinker ESP32-CAM with ESP32
Board ID:ai-thinker-esp32-cam

```
>>> import microcontroller
>>> microcontroller.cpu.frequency
240000000
>>> microcontroller.cpu.frequency=120000000
>>> microcontroller.cpu.frequency=80000000
>>>
>>> microcontroller.cpu.frequency=20000000
▒▒▒▒▒▒▒

```
Same happens if you set 40 Mhz

```
>>> import microcontroller
>>> microcontroller.cpu.frequency
240000000
>>> microcontroller.cpu.frequency=120000000
>>> microcontroller.cpu.frequency=80000000
>>>
>>> microcontroller.cpu.frequency=40000000
▒

```

dhalbert, 2025-01-18:
This is a "beware of the consequences" thing. As mentioned in https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.Processor.frequency:

> Warning
>
>Changing the frequency may cause issues with other subsystems, such as USB, PWM, and PIO. To minimize issues, set the CPU frequency before initializing other systems.

Also discussed in the original PR, #9342.

0wwafa, 2025-01-18:
> This is a "beware of the consequences" thing.

Consequences of installing circuitpython?

Because this does not happen in micropython :P

dhalbert, 2025-01-18:
MicroPython, even the latest preview build does work with 20 MHz on an identical ESP32 board (I tested with Feather ESP32 V2 and HUZZAH32.

The frequency setting logic is functionally identical in MicroPython and CircuitPython, so it may have to do with how the UART is set up.

dhalbert, 2025-01-18:
> Consequences of installing circuitpython?
> 
> Because this does not happen in micropython :P

Please don´t be snarky. If you have information that it does work in MicroPython, please post that up front.

0wwafa, 2025-01-19:
> > Consequences of installing circuitpython?
> > Because this does not happen in micropython :P
> 
> Please don´t be snarky. If you have information that it does work in MicroPython, please post that up front.

I did not mean to be. Sorry.

dhalbert, 2025-02-01:
Related to #9536, I think.