adafruit/circuitpython #9978: [info] ESP32: how to make an http request?

0wwafa, 2025-01-19:
Perhaps I am blind,
but everywhere I read I should use adafruit_requests  but there is no such module on esp32.

```
Adafruit CircuitPython 9.2.3 on 2025-01-17; ESP32 Devkit V1 with ESP32
Board ID:doit_esp32_devkit_v1
UID:4AFC21A0E74F
```

Modules:
```
>>> help('modules')
__future__        collections       max3421e          supervisor
__main__          countio           mdns              synthio
_asyncio          digitalio         memorymap         sys
_pixelmap         displayio         microcontroller   terminalio
adafruit_bus_device                 dualbank          micropython       time
adafruit_bus_device.i2c_device      epaperdisplay     msgpack           touchio
adafruit_bus_device.spi_device      errno             neopixel_write    traceback
adafruit_pixelbuf espidf            nvm               ulab
aesio             espnow            onewireio         ulab.numpy
alarm             espulp            os                ulab.numpy.fft
analogbufio       fontio            paralleldisplay   ulab.numpy.linalg
analogio          fourwire          paralleldisplaybus                  ulab.scipy
array             framebufferio     ps2io             ulab.scipy.linalg
atexit            frequencyio       pulseio           ulab.scipy.optimize
audiobusio        gc                pwmio             ulab.scipy.signal
audiocore         getpass           rainbowio         ulab.scipy.special
audiomixer        gifio             random            ulab.utils
audiomp3          hashlib           re                usb
binascii          i2cdisplaybus     rotaryio          usb.core
bitbangio         io                rtc               vectorio
bitmaptools       ipaddress         sdcardio          warnings
board             jpegio            select            watchdog
builtins          json              sharpdisplay      wifi
busdisplay        keypad            socketpool        zlib
busio             keypad_demux      ssl
canio             locale            storage
codeop            math              struct
```

prcutler, 2025-01-19:
You have to download it from the CircuitPython libraries.  It's in the official bundle at https://circuitpython.org/libraries.  You'll want the bundle for 9.x and drag and drop `adafruit_requests.mpy` to your `lib` directory.

0wwafa, 2025-01-19:
yep.. saw them 5 minutes after my question.. thanks.

dhalbert, 2025-01-19:
Please ask support questions in https://adafru.it/discord or https://forums.adafruit.com/viewforum.php?f=60. The issues queue is for bug reports or enhancement requests. THanks.