adafruit/circuitpython #9991: OverflowError after 2038 in time.localtime()

bill88t, 2025-01-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.3-1-g401295f6dc on 2025-01-21; LILYGO T-Deck (Plus) with ESP32S3
```

### Code/REPL

```python
import time
>>> time.localtime(2147483647)
struct_time(tm_year=2038, tm_mon=1, tm_mday=19, tm_hour=3, tm_min=14, tm_sec=7, tm_wday=1, tm_yday=19, tm_isdst=-1)
>>> time.localtime(2147483648)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
```

### Behavior

Yea, 1/1 of 2038 works, but not after Jan 19 of 2038.

bablokb, 2025-01-23:
Yes, this is the year 2038 bug, see https://en.wikipedia.org/wiki/Year_2038_problem

dhalbert, 2025-01-31:
Closed in favor of #2573

closed 2025-01-31 by dhalbert