adafruit/circuitpython #10002: ESP32-C3 doesn't work without CIRCUITPY_LEGACY_4MB_FLASH_LAYOUT=1 in 9.2.2 and after

dhalbert, 2025-01-26:
See problems discussed in #9201 after 9.2.2 was released. Also see #9997.

dhalbert, 2025-01-26:
This bisects to #9928, not suprising.

dhalbert, 2025-01-27:
This is connected to BLE, which is enabled when `CIRCUITPY_LEGACY_4MB_FLASH_LAYOUT` is disabled. `CIRCUITPY_BLEIO = 0` makes the problem go away. `CONFIG_COMPILER_SAVE_RESTORE_LIBCALLS` enabled or disabled doesn't make any difference. `BINARY_BLOBS` looks fine: the blob chosen is the same as before

eightycc, 2025-05-06:
Resolved boot loop with ESP-IDF 5.4.1 _and_ additional loader fragment `esp32c3.rom.eco3_bt_funcs.ld`. Build was missing ROM routine entry point addresses causing it to link some incorrect BLE code for the ESP32-C3.

Due to insufficient space in flash for BLE, I had to change to the single OTA partition map for flash to allow testing.

closed 2025-05-08 by tannewt