adafruit/circuitpython #10004: Safe Mode: Hard fault: Memory access or instruction error / Internal watchdog timer expired.

mario872, 2025-01-27:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.3-14-g52b5b2da1a on 2025-01-26; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import board
import time
import array
import pulseio

f = open("/codes.txt", "r") # https://github.com/mario872/Sonic-Screwdriver/blob/main/Software/src/codes.txt
# Just moved some lines around from https://raw.githubusercontent.com/adafruit/Adafruit_Learning_System_Guides/master/CircuitPython_TVBGone/codes.txt
input() # To make sure I can tell it when to start
count = 0
for line in f:
    code = eval(line)
    print(code)
    # If this is a repeating code, extract details
    try:
        repeat = code["repeat"]
        delay = code["repeat_delay"]
    except KeyError:  # by default, repeat once only!
        repeat = 1
        delay = 0
    # The table holds the on/off pairs
    table = code["table"]
    pulses = []  # store the pulses here
    # Read through each indexed element
    for i in code["index"]:
        pulses += table[i]  # and add to the list of pulses
    pulses.pop()  # remove one final 'low' pulse

    with pulseio.PulseOut(
        board.D13, frequency=code["freq"], duty_cycle=2**15
    ) as pulse:
        for i in range(repeat):
            pulse.send(array.array("H", pulses))
            time.sleep(delay)

    time.sleep(code["delay"])
    count += 1
```

### Behavior

Well, when running the above code I get:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

Interestingly, when I run a more complicated program, with this script basically as a function instead, I get:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.
```

### Description

N/A

### Additional information

Running a very similar script on the RPi Pico W didn't produce any error less than two weeks ago (with that build of CP)

eightycc, 2025-03-27:
`common_hal_pulseio_pulseout_deinit()` lacks a guard against calls where the `pulseio_pulseout_obj_t` is already deinit'ed. With that fixed, there's a repeating:
```
E (518334) rmt: rmt_tx_wait_all_done(595): flush timeout
I (518340) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (518448) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (518448) rmt: rmt_tx_wait_all_done(595): flush timeout
I (518452) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
```

dhalbert, 2025-03-27:
Fixed by #10190

PR #10190 (2025-03-27): Add guards to common_hal_pulseio_pulse(in|out)_deinit to prevent doub…
Add guards to `common_hal_pulseio_pulsein_deinit()` and `common_hal_pusleio_pusleout_deinit()` to prevent double release of the `pulseio_pulse(in|out)` object's resources.

Fixed same issue in  `pulseout`.

Resolves #10004.
closed 2025-03-27 by dhalbert