adafruit/circuitpython #10007: Nimble out of memory when trying to connect to multiple devices on ESP32 S3

Rutgergroos, 2025-01-27:
### CircuitPython version

```python
9.2.1
```

### Code/REPL

```python
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from cycling_power_service import CyclingPowerService
from adafruit_ble import BLERadio
import time

ble_radio = BLERadio()
advertisement = ProvideServicesAdvertisement(CyclingPowerService())
ble_radio.start_advertising(advertisement)

def advertise_ble():
    while True:
        print(ble_radio.connections)
        if len(ble_radio.connections) < 3:
            if not ble_radio.advertising:
                ble_radio.start_advertising(advertisement)
        time.sleep(2)
        
advertise_ble()

from adafruit_ble.services import Service
from adafruit_ble.characteristics import Characteristic
from adafruit_ble.attributes import Attribute
from adafruit_ble.uuid import StandardUUID

class CyclingPowerService(Service):
    uuid = StandardUUID(0x1818)

    measurement = Characteristic(
        uuid=StandardUUID(0x2A63),
        properties=Characteristic.NOTIFY,
        read_perm=Attribute.OPEN,
        write_perm=Attribute.NO_ACCESS,
        max_length=8,  
        fixed_length=True
    )
    
    feature = Characteristic(
        uuid=StandardUUID(0x2A65),
        properties=Characteristic.READ,
        read_perm=Attribute.OPEN,
        write_perm=Attribute.NO_ACCESS,
        max_length=4,
        fixed_length=True,
        initial_value=bytes([
            0x20 & 0xFF,
            (0x20 >> 8) & 0xFF,
            (0x20 >> 16) & 0xFF,
            (0x20 >> 24) & 0xFF
        ])
    )

    sensor_location = Characteristic(
        uuid=StandardUUID(0x2A5D),
        properties=Characteristic.READ,
        read_perm=Attribute.OPEN,
        write_perm=Attribute.NO_ACCESS,
        max_length=1,
        fixed_length=True,
        initial_value=(bytes([0x06]))
    )
```

### Behavior

When the first device connects everything is ok as is reported by the print. once the device starts advertising again an error is given:
(<BLEConnection object at 0x3c198930>,)
Traceback (most recent call last):
  File "<stdin>", line 18, in <module>
  File "<stdin>", line 15, in advertise_ble
  File "adafruit_ble/__init__.py", line 219, in start_advertising
MemoryError: Nimble out of memory

dhalbert, 2025-01-27:
It looks like you are allowing three connections to be established at once. Even after a connection is established you start advertising again. Is that your intention?

Rutgergroos, 2025-01-27:
Yes that is correct. this is for a bike computer that should connect to a watch and a zwift server
also just tested in 9.2.3 with latest libraries and i see the same behaviour. 

strange thing is that sometimes i can connect 2 devices but most of the time its only 1

Rutgergroos, 2025-01-27:
i have also tried something similar to the examples

    ble.stop_advertising()  # you need to do this to stop any persistent old advertisement
    ble.start_advertising(advertisement)

same error will still occur

dhalbert, 2025-01-27:
Which board are you using?

Rutgergroos, 2025-01-28:
Hi, i have tested this on TTGO T-Display-S3 1.9 (16m8m) and the ESP32-S3-DevKitC-1-N8R2

the snippet above is an oversimplified version of what i am running for the total bike computer. With the interface and everything active it uses about 170kb (for TTGO) vs the 60kb (DevKitC) that this sketch is using. 

sometimes the device allows me to connect to and then crashing at a third connection and sometimes i can just connect one device

i was using gc.mem_free() to check the memory usage is this applicable to nimble or does it have its own memory allocated?

JoeyWangTW, 2025-08-17:
Hi I'm also running into similar issue when trying to run [KMK](https://github.com/KMKfw/kmk_firmware) on Seeed XIAO ESP32C3. I can sometimes successfully connect and working, but it disconnects after a few seconds. I'm seeing this error message
```
  File "adafruit_ble/__init__.py", line 212, in start_advertising
MemoryError: Nimble out of memory
```

dhalbert, 2025-11-11:
If the "crashes" are all `Nimble out of memory`, then the issue is just that the storage allocated for BLE is insufficient for both the workflow and your program.

bbilger, 2025-11-11:
Right. But the unexpected part - at least for me - was that the _BLE workflow_ automagically starts at all just because there's some bonded device.

dhalbert, 2026-09-03:
This may also be fixed or ameliorated by #11304.