adafruit/circuitpython #10024: RP2040 rotaryio fails with two encoders

todbot, 2025-02-03:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4 on 2025-01-29; Raspberry Pi Pico with rp2040

and:
Adafruit CircuitPython 9.2.4-5-g3236a0f200 on 2025-01-31; Raspberry Pi Pico with rp2040  
  ("absolute newest" as of noon PST 3 Feb 2025)
Adafruit CircuitPython 9.2.3 on 2025-01-17; Raspberry Pi Pico with rp2040
Adafruit CircuitPython 9.2.2 on 2025-01-09; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import time, board, rotaryio

encoder1 = rotaryio.IncrementalEncoder(board.GP19, board.GP18)
encoder2 = rotaryio.IncrementalEncoder(board.GP4, board.GP5)

while True:
    print(encoder1.position, encoder2.position)
    time.sleep(0.1)
```

### Behavior

Code hangs on creation of `encoder2`, which is obviously not correct.

Code should repeatedly print position of both encoders.

### Description

Works on 9.2.1. 
Does not work on 9.2.2 and above. 
Related to #9682? 

### Additional information

Redditor had his issue too: https://old.reddit.com/r/circuitpython/comments/1igqzyf/two_incrementalencoders_fail/

todbot, 2025-02-03:
Update: It appears to get out of hanging if a change happens on `encoder2`.  But then `encoder1` does not work.

jepler, 2025-02-03:
From 9.2.1 to 9.2.2 the big thing seems to be my PIO re-org to support the high pins on rp2350b (#9901). I know there were also changes to interrupts (or was it DMA? which this driver doesn't use-- #9980) but those seem to have occurred after 9.2.3.

jepler, 2025-02-04:
I wrote some code so that I didn't need any encoder knobs for testing.

```py
# Test simulated encoder (issue #10024) for metro rp2350
# Connect jumper wires from RX/TX/D2/D3 to A0/A1/A2/A3
# Each encoder should count.
import board, rotaryio
import digitalio

d = digitalio.DigitalInOut(board.D0)
d.switch_to_output(False)
e = digitalio.DigitalInOut(board.D1)
e.switch_to_output(False)
f = digitalio.DigitalInOut(board.D2)
f.switch_to_output(False)
g = digitalio.DigitalInOut(board.D3)
g.switch_to_output(False)

r1 = rotaryio.IncrementalEncoder(board.A0, board.A1, divisor=1)
r1.position = 0
r2 = rotaryio.IncrementalEncoder(board.A2, board.A3, divisor=1)
r2.position = 0

for _ in range(8):
    d.value = True
    print(f"{+d.value}{+e.value} {r1.position} {r2.position}")
    e.value = True
    print(f"{+d.value}{+e.value} {r1.position} {r2.position}")
    d.value = False
    print(f"{+d.value}{+e.value} {r1.position} {r2.position}")
    e.value = False
    print(f"{+d.value}{+e.value} {r1.position} {r2.position}")

for _ in range(8):
    f.value = True
    print(f"{+f.value}{+g.value} {r1.position} {r2.position}")
    g.value = True
    print(f"{+f.value}{+g.value} {r1.position} {r2.position}")
    f.value = False
    print(f"{+f.value}{+g.value} {r1.position} {r2.position}")
    g.value = False
    print(f"{+f.value}{+g.value} {r1.position} {r2.position}")
```

todbot, 2025-02-04:
Verified that fix https://github.com/adafruit/circuitpython/pull/10025 does make two encoders work on RP2040 Pico and RP2350 Pico2.

jepler, 2025-02-04:
Thanks for testing @todbot !

PR #10025 (2025-02-04): raspberrypi: Fix state machine allocation
.. for duplicate programs, like two IncrementalEncoder instances.

Closes: #10024

Also removes a workaround that's no longer needed after a pico sdk update.