adafruit/circuitpython #10030: Audible stepping when using `synthio.LFO` on `audiodelays.Echo.delay_ms`

relic-se, 2025-02-06:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4; Pimoroni Pico Plus2 with rp2350b
```

### Code/REPL

```python
import audiobusio
import audiodelays
import board, time
import synthio

SAMPLE_RATE = 48000
AMPLITUDE = 0.2
MIX = 1.0 # Would need to be 0.5 if #9994 is merged
TIME = 40
DEPTH = 8
RATE = 1.0
OCTAVE = -1

audio = audiobusio.I2SOut(
    bit_clock=board.GP0,
    word_select=board.GP1,
    data=board.GP2,
)

synth = synthio.Synthesizer(
    sample_rate=SAMPLE_RATE,
    channel_count=2,
)

effect = audiodelays.Echo(
    max_delay_ms=TIME,
    delay_ms=synthio.LFO(offset=TIME, scale=DEPTH, rate=RATE),
    decay=0.0,
    sample_rate=SAMPLE_RATE,
    channel_count=2,
    freq_shift=True,
)

effect.play(synth)
audio.play(effect)

# Chord
for i in range(3):
    synth.press(synthio.Note(
        frequency=synthio.midi_to_hz(65 + i * 4 + OCTAVE * 12),
        panning=synthio.LFO(rate=0.15 + i * 0.05),
        amplitude=AMPLITUDE,
    ))

while True:
    if effect.mix:
        effect.mix = 0.0
        print("Off")
    else:
        effect.mix = MIX
        print("On")
    time.sleep(2.0)
```

### Behavior

There is audible stepping in the chorus effect.

### Additional information

The optimization at Echo.c#L313 prevents calling `recalculate_delay` at it's maximum rate of `SYNTHIO_MAX_DUR` and instead limits it to increments of `sample_ms`. For effects using larger delay lengths, this stepping isn't noticeable, but for effects which require a delay with a shorter delay length (ie: chorus, flanger), more granularity is required.

https://github.com/adafruit/circuitpython/blob/4ba872cae51be6caf7ce80b082a15e376ba4a036/shared-module/audiodelays/Echo.c#L313-L315

I'm thinking that a suitable solution is to utilize the method, `float_equal_or_update`, provided within `synthio.BlockBiquad` to check if the delay needs to be recalculated. That, or check the difference by `sample_ms / 256` (for the 8-bits of sub-resolution). This additional granularity may only be necessary if `freq_shift=True`.

https://github.com/adafruit/circuitpython/blob/4ba872cae51be6caf7ce80b082a15e376ba4a036/shared-module/synthio/BlockBiquad.c#L65-L74

relic-se, 2025-02-06:
I've tested the proposed fix, and it doesn't appear to resolve the issue. I've also toyed around with a few other adjustments to no avail. Any ideas as to why this could be happening would be greatly appreciated.

d1ffeq-diy, 2025-02-10:
Don't have the 2350 board at the moment, can you try to replace your chord lines with a single tone note, like this:
`synth.press(synthio.Note(frequency=440, amplitude=AMPLITUDE))`

to narrow things down, as I've noticed in your sound example that stepping occurs approx. every 4.2s, exactly the period of your panning parameter LFO.

relic-se, 2025-02-11:
@d1ffeq-diy Thank you for the review. While setting up an updated example to simplify the demonstration, I think I've come across the problem, and it's much simpler than I expected.

As per the current documentation on `synthio.LFO` (https://docs.circuitpython.org/en/latest/shared-bindings/synthio/index.html#synthio.LFO), it states "If `waveform` is None, a triangle waveform is used." When I specify the waveform directly (such as a sine wave in the example below), the program works as intended.

```
import audiobusio
import audiodelays
import audiofilters
import board, time
import ulab.numpy as np
import synthio

WAVEFORM = True # False is default, True is sine
MIX = 1.0 # Would need to be 0.5 if #9994 is merged

audio = audiobusio.I2SOut(
    bit_clock=board.GP0,
    word_select=board.GP1,
    data=board.GP2,
)

synth = synthio.Synthesizer(
    sample_rate=48000,
    channel_count=1,
)

effect_delay = audiodelays.Echo(
    max_delay_ms=50,
    delay_ms=synthio.LFO(
        offset=40,
        scale=10,
        rate=1.0,
        waveform=None if not WAVEFORM else np.array(np.sin(np.linspace(0, 2*np.pi, 512, endpoint=False)) * 32767, dtype=np.int16),
    ),
    decay=0.0,
    sample_rate=48000,
    channel_count=1,
    freq_shift=True,
)

effect_delay.play(synth)
audio.play(effect_delay)

synth.press(synthio.Note(
    frequency=synthio.midi_to_hz(53),
    amplitude=0.5,
))

while True:
    if effect_delay.mix:
        effect_delay.mix = 0.0
        print("Off")
    else:
        effect_delay.mix = MIX
        print("On")
    time.sleep(2.0)
```

Audio recording of the above example, first with `WAVEFORM=False` then with `WAVEFORM=True`: https://drive.google.com/file/d/1SbZjiso4bfjI0rYGKnfCFdzwM7BZ7IuZ/view?usp=sharing

I'll need to double check the constructor code for `synthio.LFO`, but it's possible that it's actually defaulting to a square wave of sorts. However, I don't remember it exhibiting this functionality in the past.

relic-se, 2025-02-11:
If I replace `time.sleep(2.0)` in the loop with the following:

```
tick = 40
while tick:
    print(effect_delay.delay_ms.value)
    time.sleep(0.05)
    tick -= 1
```

I can generate the graph below (in LibreOffice Calc):

[image]

It looks as if the default waveform is functioning as intended, but it is somehow not affecting the effect parameter correctly unless a waveform is specified by the user.

d1ffeq-diy, 2025-02-11:
There's definitely a bug. Looking closely at harmonic content in spectrogram of the latest audio snippet, the modulation with `WAVEFORM=False` looks square-like: 
[image]
and with `WAVEFORM=True` it looks exactly as it should: 
[image]

also, the tiny slopes of the square-like transitions suggest that it _once was_ a triangle wave. My theory is that default LFO values do not scale properly somewhere further down the line, causing waveform to be heavily clipped(amplified too much) between min and max values.

d1ffeq-diy, 2025-02-23:
@relic-se any progress on tracking the LFO creation code?

relic-se, 2025-02-24:
@d1ffeq-diy I haven't dig too deep into it as of yet. If you've got the know-how, feel free to poke around in the core. Otherwise, because the work-around is fairly simple (just providing your own waveform), I've got this as a low priority item atm.

mikeysklar, 2026-11-02:
The jump looks expected rather than a bug.

With `freq_shift=True`, pitch follows how fast the delay is changing, so a triangle LFO gives two steady pitches with a jump between them.

At 1 Hz and 40 +/- 10 ms, I measured +69.5 and -69.0 cents, against a predicted +/-69.3. A sine LFO should glide instead.

There is also a smaller real issue: the rate only has 8 fractional bits, which adds pitch blips of about 5 cents on slow LFOs.

| Slow sine, 0.1 Hz, 40 +/- 1 ms | Now | 12 bits |
|---|---|---|
| Error, cents rms | 1.19 | 0.54 |
| Peak, cents | 5.09 | 3.53 |

The 12-bit patch costs 136 bytes on a Metro RP2350 and limits very long buffers, so 192 kHz tops out at 2730 ms.

It does not remove the triangle jump, and nobody has listened to it yet.

Would you like that as a PR, with a docs note about the triangle behavior?

Measured with Claude on the Unix build, not on a board.

closed 2026-11-11 by dhalbert