adafruit/circuitpython #10031: rp2350: dvi display blanks during flash writes

jepler, 2025-02-06:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.3-17-gfe23bd7cf8-dirty on 2025-02-04; Adafruit Metro RP2350 with rp2350b
```

### Code/REPL

```python
import board
import picodvi
import displayio
import framebufferio

displayio.release_displays()

fb = picodvi.Framebuffer(320, 240, clk_dp=board.CKP, clk_dn=board.CKN,
                                   red_dp=board.D0P, red_dn=board.D0N,
                                   green_dp=board.D1P, green_dn=board.D1N,
                                   blue_dp=board.D2P, blue_dn=board.D2N,
                                   color_depth=16)
display = framebufferio.FramebufferDisplay(fb)
```

### Behavior

After the code runs, the display is visible. However, copying a file to CIRCUITPY causes the display to blank. It seems the display is not outputting anything at all (not correct sync), because my [HDMI to USB video capture dongle](https://www.adafruit.com/product/4669) shows a test pattern during this time. After the write finishes, the display comes back.

### Description

This happens back to 9.2.0, and both on a Feather RP2350 with self-soldered PSRAM and on a Metro RP2350 prototype. Encountered while working on #10028 but is not caused by the method that PR uses to set up the display.

jepler, 2025-02-06:
It looks like picodvi on rp2350 uses an IRQ on core0 to start the frame via `irq_set_exclusive_handler(DMA_IRQ_1, dma_irq_handler);` in `common_hal_picodvi_framebuffer_construct` which is running in core0. So, it's expected that flash writing blocks this irq. we'd need to place the irq handler on core1 for this to work. or else not fully block this specific interrupt during flash writes, as it should be safe.

tannewt, 2025-02-06:
Looks like `common_hal_mcu_disable_interrupts` guards flash writes:

https://github.com/adafruit/circuitpython/blob/4ba872cae51be6caf7ce80b082a15e376ba4a036/ports/raspberrypi/supervisor/internal_flash.c#L116-L135

We could have it mask by priority and set the DMA interrupt so it is higher. We need to ensure it can run without flash then though.

I don't think running on core1 helps because we're trying to avoid accessing flash by disabling interrupts.

PR #10049 (2025-02-12): raspberrypi: Don't block DMA_IRQ_1 during common_hal_mcu_disable_interrupts
doing so causes the picodvi display on rp2350 to blank, because this irq is handed by cpu0.

Instead, use the BASEPRI register so that common_hal_mcu_disable_interrupts masks interrupts with lower priority (higher priority values) than PICO_ELEVATED_IRQ_PRIORITY. This has the effect of masking "regular" interrupts while still letting this priority interrupt occur.

port_idle_until_interrupt now needs to directly manipulate the interrupt enable state, because an interrupt masked via BASEPRI also does not cause the WFI i [... cut]
closed 2025-02-12 by jepler