adafruit/circuitpython #10032: mpy_cross

schaefer01, 2025-02-06:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.0.5 on 2024-05-22 Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
ValueError: incompatible .mpy file
```

### Behavior

#
# mit haystack observatory
# rps 2/6/2025
#
# done in the attempt to save memory space for execution
#
# $ python3 -m mpy_cross ~/eclipse/gps/staging/gnss_imu_mag.py
#
import gnss_imu_mag as gim
gim.my_main()

### Description

I'm looking for a version of mpy_cross that works with circuitpython v9
the micropython's github is version 6.3 and claim no responsibility for upgrading the mpy_cross tool

$ python3 -m mpy_cross --version
  MicroPython v1.24.1 on 2024-11-30; mpy-cross emitting mpy v6.3

your github source does not respond well to make:

$ make

- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
Cannot determine version.

CircuitPython must be built from a git clone with tags.
If you cloned from a fork, fetch the tags from adafruit/circuitpython as follows:

$    make fetch-tags
make: *** [build/genhdr/mpversion.h] Error 1
rps@C02D81GJMD6W mpy-cross % make fetch-tags
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
make: *** No rule to make target `fetch-tags'.  Stop.

### Additional information

Not sure what to do other than, per this: https://forums.adafruit.com/viewtopic.php?t=208055

suggesting I should go back to circuitpython v8

please advise,
  thank you
    bob s

dhalbert, 2025-02-06:
Find a recent version here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/ 
Look in the operating-system-name directories at the top for a 9.2.x version.

mpy-cross doesn't change much. Any 9.x.x version should be ok.

schaefer01, 2025-02-06:
thank you, I'll close after I get it running

schaefer01, 2025-02-06:
this works, thank you.

The URL that I initially used for the circuitpy "welcome" was not the same as the one in dhalbert's reply, 
 that one had a broken link, with led to googling and finding many dead ends. 
However, its all good in the end.
thank you!!!

closed 2025-02-06 by schaefer01

schaefer01, 2025-02-06:
undo!  mpy-cross from the URL in the previous reply is still out of date

I tested against the .py file, not the .mpy file

./mpy-cross-macos-11-8.0.5-x64 ./gnss_imu_mag.py

the build runs but the import returns the same error:

code.py:
#
# mit haystack observatory
# rps 2/6/2025
#
# done in the attempt to save memory space for execution
#
# $ python3 -m mpy_cross ~/eclipse/gps/staging/gnss_imu_mag.py
#
import gnss_imu_mag as gim
gim.my_main()

the rp2040:
0;?🐍Done | 9.0.5soft reboot

Auto-reload is off.
code.py output:
0;?🐍code.py | 9.0.5Traceback (most recent call last):
  File "code.py", line 9, in <module>
ValueError: incompatible .mpy file
0;?🐍9@code.py ValueError | 9.0.5
Code done running.

schaefer01, 2025-02-06:
hi,
 the date of the mpy_cross file is 2023, circuitpy v8 was still prevalent 2 years ago
 the "forums" link in my first post pointing out the problem was dated from february a year ago.

dhalbert, 2025-02-06:
Please look inside these directories, as listed below and at the top of
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/

The 8.x.x builds at the top leve are out of date, yes, but don't use those:

[../](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/)
                                0              [linux-aarch64/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/linux-aarch64/)
                                0              [linux-amd64/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/linux-amd64/)
                                0              [linux-raspbian/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/linux-raspbian/)
                                0              [macos-11/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/macos-11/)
                                0              [macos/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/macos/)
                                0              [windows/](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/mpy-cross/windows/)
[0_Look_in_the_operating_system_directories_above_for_the_latest_versions.README](https://adafruit-circuit-python.s3.amazonaws.com/bin/mpy-cross/0_Look_in_the_operating_system_directories_above_for_the_latest_versions.README)

schaefer01, 2025-02-06:
thank you again, I'm all set for sure this time

  this is what happens when I work in panic mode before taking a break for lunch lunch

dhalbert, 2025-02-06:
I try not to post before breakfast, same idea :)