adafruit/circuitpython #10042: ESP32-C6-LCD-1.47 LDO overheat

peterzhenghj, 2025-02-08:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-C6 LCD 1.47 with ESP32-C6FH4
```

### Code/REPL

```python
# NO code to run
```

### Behavior

After the flash restart, I noticed that the area of the low dropout regulator (LDO) feels overheated when touched. 
However, I only have one unit available, so I am not sure if this issue is unique to my device. 
Interestingly, when I flash the factory images, there is no overheating situation.

### Additional information

[Waveshare product wiki](https://www.waveshare.com/wiki/ESP32-C6-LCD-1.47)

kylefmohr, 2025-02-09:
I was able to recreate this (any excuse to use my thermal camera!) but I question whether this is specifically a CircuitPython issue, or if this more relates to the fact that [Python is an interpreted language](https://en.wikipedia.org/wiki/Interpreter_(computing)#Efficiency). I would theorize that you would see the same amount of heat if you created a program in C/C++/Arduino's language which was always running at 100%.

Nevertheless, here are the temps I found.

First with the factory image loaded, I saw it max out at about 107°F (~42°C), and the LCD capped out around 92°F (~33°C).

[image] [image]

Then once I flashed CircuitPython onto the device, it quickly climbed to 140°F (~60°C), as did the LCD screen.

[image]  [image]

and after leaving it idle for a few hours (Just waiting at the REPL prompt for CircuitPython), the device was at about 160 F (71 C)

[image]

<details><summary>Here's a video of the temperature climbing as the device is plugged into power</summary>
<p>

https://github.com/user-attachments/assets/be0774c3-e83f-446f-afc2-0e1f8700028e

</p>
</details> 

But I will just say one more time, I don't think CircuitPython is to blame for this.

tannewt, 2025-02-10:
Please talk to Waveshare about this. Maybe there is some pin state that needs to be set in CircuitPython.

peterzhenghj, 2025-02-13:
> Please talk to Waveshare about this. Maybe there is some pin state that needs to be set in CircuitPython.

Yes I will, thank you sir.

dhalbert, 2025-07-29:
@peterzhenghj did you hear anything back from Waveshare?

effesoftwarecom, 2025-08-22:
Overheating problem is also noticed in my case. With a basic LVGL application with only a spinner, the part of screen near the USB connector is getting really hot. In fact, screen in this area is getting black, making the gadget unusable.

[image]

apws, 2026-01-17:
hi guys, just testing on this board arduino_gfx helloworld demo bliting text at max rate and I found the same thing, probably due to digtal 1 so always ON backlit, after seting this pin to pwm at 50% or so, everything looks good; cheers;