adafruit/circuitpython #10045: Standalone CircuitPython Computer Ideas

tannewt, 2025-02-08:
Here are some random ideas to make CircuitPython better standalone.
* Add hot key to switch between terminal display and program display.
* Add program suspend state that is stored in RAM so that switching between programs doesn't need to lose all state. For example, a code editor may want to save the open file(s) and line numbers. It wouldn't be stored in flash.
* Add program browser to "activate" or "install" programs into slots for F1-F12 keys. Serves as a quick way to switch between them. Could fetch programs from github repo.
* Add foreground and background color support per-tile in TileGrid. This would allow terminal coloring and color inversion.
* Add emoji support. https://emoji.serenityos.org/
* Support gamepad switching too.
* LVGL displayio for speed? Especially scrolling terminal.
* `keypad`-like buffer for USB HID devices?
* Software
  * Code editor
  * Tracker
  * Image editor. w/tile guides
  * Pong
  * Snake

jepler, 2025-02-10:
what if we partitioned the fruit jam so that it has a CIRCUITPY filesystem area (writable by host) plus a second area that is not on USB so it's always writable by code, and is automounted. `/local` or so.

eightycc, 2025-02-10:
Maybe some classic computer emulation? Generic CP/M, Apple ][, perhaps a minicomputer like the PDP-8?

tannewt, 2025-02-10:
> what if we partitioned the fruit jam so that it has a CIRCUITPY filesystem area (writable by host) plus a second area that is not on USB so it's always writable by code, and is automounted. `/local` or so.

We could... or is /sd automounting enough?

> Maybe some classic computer emulation? Generic CP/M, Apple ][, perhaps a minicomputer like the PDP-8?

I don't think we would from within CircuitPython. It could be another image/Learn guide but not CircuitPython based anyway.