adafruit/circuitpython #10047: Installing Circuitpython on RPi pi pico 2 W puts uf2 file in filesystem, doesn't reboot

awildenberg, 2025-02-09:
### CircuitPython version and board name

```python
Attempting to install CircuitPython 9.2.4 on new Raspberry Pi Pico 2W

INFO_UF2.TXT contains 
 Bootloader v3.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
```

### Code/REPL

```python
problem is in installation, so no relevant Code to share.
```

### Behavior

- Board always seems to be in BOOTSEL mode.
- Board doesn't seem to recognize/respond to adafruit-circuitpython-raspberry_pi_pico2_w-en_US-9.2.?.uf2
- - file appears in device's RPI-RP2 directory
- - file disappears after manual reboot
- Board seems to recognize and respond to flash_nuke.uf2

### Description

I've done quite a bit of circuitpython development building MQTT based temperature systems involving quite a few board/sensor combinations, including raspberry pi pico 1 W boards.

I have several raspberry pi pico 2 W boards that i recently bought, and i can't get any of them to install circuitpython.  Running, Mac OS X 15.2 (Sequoia).  Here's what I'm doing:
1) Hold BOOTSEL down
2) plug in RPI
3) wait for /Volumes/RPI-RP2 to appear
4) release BOOTSEL
5) "copy" adafruit-circuitpython-raspberry_pi_pico2_w-en_US-9.2.?.uf2 to /Volumes/RPI-RP2
6) uf2 file appears when I do ls /Volumes/RPI-RP2
7) device doesn't reboot (I've waited short and long times).

where "copy" is one of a) drag-and-drop in finder, b) cp command, c) rsync command, and '?' is 3 or 4.

If I manually disconnect and reconnect, /Volumes/RPI-RP2 appears, but the uf2 file is now gone.

If I do the same thing but copy flash_nuke.uf2 over, the device immediately reboots, disconnecting and when it comes back I see the default /Volumes/RPI-RP2 files.

trying screen /dev/tty*usb* there's no such device.

Interestingly enough, if I *don't* hold down BOOTSEL, I get exactly the same behavior for all of this (including nuke_flash.uf2 rebooting things).  I would have guessed maybe there's a board defect that makes BOOTSEL never work, but the fact that flash_nuke works on the board seems to indicate that's not likely.

dhalbert, 2025-02-09:
> 6. uf2 file appears when I do ls /Volumes/RPI-RP2

If the UF2 file name appears in `RPI-RP2`, that is an indication that the upload failed. Normally it would not. This could be for a variety of reasons, including:
- Operating system copying failure. What version of macOS are you running? I have seen one report of issues with 15.3. If you have another computer (Windows, Linux, RPi Linux SBC, etc.), try with one of those.
- UF2 file is corrupted. Re-download and try again.

Try loading the plain Pi Pico 2 UF2 as well.

RetiredWizard, 2025-02-09:
Are you sure you have a Pico 2W ? When I'm in boot select mode, the volume label is RP2350. The board should say Pico 2 W with the 2 in a bold white box. If you do have the 2W model then I'm guessing you have the wrong boot loader installed somehow.

awildenberg, 2025-02-09:
Terribly embarrassed.  Yes, in fact I'd misread the labels and pulled out a pair of Pico Ws instead of a pair of Pico 2Ws.  Running the correct UF2 did the trick.  Mortified.

Thank you for such quick responses.

dhalbert, 2025-02-09:
No problem!

closed 2025-02-09 by dhalbert