adafruit/circuitpython #10051: displayio color converter incorrect

qutefox, 2025-02-11:
### CircuitPython version and board name

```python
I have rewritten the displayio, vectorio and adafruit_imageload code as a seperate c++ codebase with tests. So if you think this "version" should be ignored you could just close this bug report.
```

### Code/REPL

```python
# Well my test could be reproduced with something similar.
# Images are from: https://entropymine.com/jason/bmpsuite/bmpsuite/html/bmpsuite.html

import displayio
import adafruit_imageload

# bmp image load test description:
# A 16-bit image with the default color format: 5 bits each for red, green, and blue, and 1 unused bit.

imageBmp, pixelShaderBmp = adafruit_imageload.load(
    "images/g/rgb16.bmp", bitmap=displayio.Bitmap, palette=displayio.ColorConverter
)

imagePng, pixelShaderPng = adafruit_imageload.load(
    "images/g/rgb16.png", bitmap=displayio.Bitmap, palette=displayio.ColorConverter
)

for y in range(imageBmp.height):
    for x in range(imageBmp.width):
        bmpValue = pixelShaderBmp.convert(imageBmp[x, y])
        pngValue = pixelShaderPng.convert(imagePng[x, y])
        if bmpValue != pngValue:
            print("got pixel (in the bmp image) does not match the expected pixel (in the png image)")
```

### Behavior

I get non-matching pixel messages.
(See the description for the explanation, and additional information for the "fix".)

### Description

The bmp image that is being loaded has RGB555 color space. Inside the imageload module the color is converted like this:
``` bitmap_obj[offset + x] = converter_obj.convert(pixel) ```
https://github.com/adafruit/Adafruit_CircuitPython_ImageLoad/blob/main/adafruit_imageload/bmp/truecolor.py#L129
where the converter_obj is created like this:
``` input_colorspace = Colorspace.RGB555 ```
https://github.com/adafruit/Adafruit_CircuitPython_ImageLoad/blob/main/adafruit_imageload/bmp/truecolor.py#L90

This conversion will happen here:
https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/ColorConverter.c#L318
Where the rgb555 color is first converted into rgb888, then the rgb888 is converted into rgb565.
The problem happens during the first conversion here:
```
case DISPLAYIO_COLORSPACE_RGB555: {
            uint32_t r8 = (pixel >> 10) << 3;
            uint32_t g8 = ((pixel >> 5) << 3) & 0xff;
            uint32_t b8 = (pixel << 3) & 0xff;
            pixel = (r8 << 16) | (g8 << 8) | b8;
        }
```
https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/ColorConverter.c#L199

If the rgb555 input color is: 0x7e10 then the calculated rgb888 is 0xf88080 instead of the correct 0xff8484. And from here (rolling the error into the 2nd conversion) the rgb565 will be 0xfc10 instead of the correct 0xfc30.

### Additional information

Based on this article: https://stackoverflow.com/questions/71106056/what-is-the-correct-way-to-convert-rgb555-to-rgb888
I changed the above code to:
```
case ColorSpace::COLORSPACE_RGB555:
    {
        uint8_t r8 = (pixel >> 10) & 0x1f;
        uint8_t g8 = (pixel >> 5) & 0x1f;
        uint8_t b8 = pixel & 0x1f;
        r8 = (r8 << 3) | ((r8 >> 2) & 0b111);
        g8 = (g8 << 3) | ((g8 >> 2) & 0b111);
        b8 = (b8 << 3) | ((b8 >> 2) & 0b111);
        pixel = (r8 << 16) | (g8 << 8) | b8;
    }
```
In file: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/ColorConverter.c#L199

and it solved the problem for me. After the change my test passed.
I hope my finding is good and valid and a little bit appreciated.

tannewt, 2025-02-13:
Thank you for the thorough issue. Would you mind filing a pull request with this change? I bet the other conversions to RGB888 have the same issue. (RP2350 HSTX has this issue too.)

qutefox, 2025-02-17:
I get

"git.exe push --set-upstream --progress "origin" fix-displayio_colorconverter_555-main
remote: Permission to adafruit/circuitpython.git denied to qutefox.
fatal: unable to access 'https://github.com/adafruit/circuitpython/': The requested URL returned error: 403"

when I try to push my changes into a new branch started from main. I don't know what I am doing wrong.
Here is the patch file: 

[0001-fix-displayio-ColorConverter-was-inaccurate-when-con.patch](https://github.com/user-attachments/files/18821641/0001-fix-displayio-ColorConverter-was-inaccurate-when-con.patch)

Neradoc, 2025-02-17:
That's because you must work in your fork and PR from there:
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github/grab-your-fork

qutefox, 2025-02-17:
Made PR: https://github.com/adafruit/circuitpython/pull/10076

tannewt, 2025-02-18:
Thank you for #10076

closed 2025-02-18 by tannewt