adafruit/circuitpython #10053: Feather ESP32-V2 and MAX3241 Featherwing

mikeysklar, 2025-02-12:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; Adafruit Feather ESP32 V2 with ESP32
```

### Code/REPL

```python
import time
import board
import max3421e
import usb

spi = board.SPI()
cs = board.D33
irq = board.D15

host_chip = max3421e.Max3421E(spi, chip_select=cs, irq=irq)

while True:
    print("Finding devices:")
    for device in usb.core.find(find_all=True):
        # pylint: disable=line-too-long
        print(f"{device.idVendor:04x}:{device.idProduct:04x}: {device.manufacturer} {device.product}")
    time.sleep(5)
```

### Behavior

Both CircuitPython 9.2.1 and the current 9.2.4 fail to discover a microSD adapter plugged into the MAX3421E when using a Feather V2. This is using the guide example USB host scan code.

This works with Arduino.

This works with (irq/cs pin changes) using CircuitPython 9.2.4 on a Feather RP2040 DVI.

### Description

Working Feather RP2040 DVI

```
Adafruit CircuitPython 9.2.4 on 2025-01-28; Adafruit Feather RP2040 DVI with rp2040
Finding devices:
0781:cfcb: SanDisk SDDR-B531
Finding devices:
Finding devices:
0781:cfcb: SanDisk SDDR-B531
```

Feather V2 (CP 9.2.1, 9.2.4)
```
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
```

Arduino with same Feather V2
```
+---------------------------+---------+
| Component                 | Version |
+---------------------------+---------+
| macOS                     | 15.2    |
| Arduino IDE               | 2.3.4   |
| Adafruit TinyUSB Library  | 3.4.2   |
| SPI Library               | 3.1.1   |
+---------------------------+---------+
```

```
TinyUSB Dual: Device Info Example with MAX3421E
Device 1: ID 0781:cfcb
Device Descriptor:
  bLength             18
  bDescriptorType     1
  bcdUSB              0210
  bDeviceClass        0
  bDeviceSubClass     0
  bDeviceProtocol     0
  bMaxPacketSize0     64
  idVendor            0x0781
  idProduct           0xcfcb
  bcdDevice           2920
  iManufacturer       3     SanDisk
  iProduct            4     SDDR-B531
  iSerialNumber       2     052522000003619
  bNumConfigurations  1
Device 1: ID 0781:cfcb SanDisk SDDR-B531
```

### Additional information

[forum issue](https://forums.adafruit.com/viewtopic.php?t=216331)

RetiredWizard, 2025-02-15:
I'm seeing the same issue using a Feather ESP32-S2 and a Feather ESP32-S3 4m/2m. I think I got  the Arduino simple example to detect occasionally but it only seemed to happen when I picked up the feather double so I think perhaps some signal noise may have been involved.

I did get a back to back QtPy ESP32-S2/USB host BFF working properly, however using jumper leads to connect the BFF to the Feather boards did not work. I also tried a QtPy ESP32-C3 and that had more serious problems (CircuitPython went to safe mode and Arduino boot looped). One interesting (and perhaps expected) observation was that when using the Arduino example, a device could only be detected once. i.e. If I plugged in a keyboard, all good, unplug, detected, plug in mouse, detected, unplug, detected, plug in original keyboard, not detected.

@mikeysklar Is there any chance you have a Feather S2 or S3 you could test against, would love to confirm this isn't something on my end.

mikeysklar, 2025-02-15:
I can hookup a Feather S2/S3 to confirm the behavior.

mikeysklar, 2025-02-16:
Feather ESP32-S2 Rev TFT
CircuitPython 9.2.4

It felt slow and unstable, but this controller did see the connected microSD card (briefly). Notice the Finding devices initially didn't pick it up over the first few passes, but then it did discover it.

```
Adafruit CircuitPython 9.2.4 on 2025-01-29; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
0781:cfcb: SanDisk SDDR-B531
Finding devices:
0781:cfcb: SanDisk SDDR-B531
Finding devices:
0781:cfcb: SanDisk SDDR-B531
Finding devices:
0781:cfcb: SanDisk SDDR-B531
```

I reset the controller (reset button) and eve with multiple removal and inserts of the USB-A plug it failed to pick it up the same card.

```
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
```

dhalbert, 2025-03-27:
Update TinyUSB and then retest.

mikeysklar, 2025-03-28:
@dhalbert I tried CircuitPython 9.2.6 (uploaded the .bin file to a Feather v2 through Web ESPTool). Would that include TinyUSB or should I be installing it differently on this board?

The Feather v2 + Max3241 was not able to detect my microSD card.

```
Adafruit CircuitPython 9.2.6 on 2025-03-23; Adafruit Feather ESP32 V2 with ESP32

Finding devices:
Finding devices:
Finding devices:
Finding devices:
```

mikeysklar, 2025-03-28:
Also tried a Feather ESP32-S2 TFT + MaX3241 with current bootloader 0.21.0 and CircuitPython 9.2.6. It doesn't print finding devices, but the code executes.

```
Adafruit CircuitPython 9.2.6 on 2025-03-23; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

```

dhalbert, 2025-03-28:
@mikeysklar Sorry for the cryptic comment about updating TinyUSB. This was part of an issue triage. We mean that we need to try updating the TinyUSB source code, which has some bug fixes, and then re-test.

mikeysklar, 2025-04-01:
CircuitPython 9.2.7 looks good. 

Feather ESP32-S2 Rev TFT + MAX3241 FW detected a microSD without issue. 

```
Adafruit CircuitPython 9.2.7 on 2025-04-01; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
[...]
code.py output:
Finding devices:
0781:cfcb: SanDisk SDDR-B531
```

mikeysklar, 2025-04-01:
No luck on the Feather v2.

```
🐍192.168.0.139 | BLE:Off | Done | 9.2.7\]0;🐍192.168.0.139 | BLE:Off | REPL | 9.2.7\
Adafruit CircuitPython 9.2.7 on 2025-04-01; Adafruit Feather ESP32 V2 with ESP32

Finding devices:
Finding devices:
Finding devices:
Finding devices:
Finding devices:
```

dhalbert, 2025-05-19:
Retest with latest 10.0.0 alpha.

mikeysklar, 2025-05-19:
I'm seeing the same behavior with CP 10.0.0-alpha6 as CP 9.2.7 on the ESP32-V2 + MAX3241. 

```
Adafruit CircuitPython 10.0.0-alpha.6 on 2025-05-17; Adafruit Feather ESP32 V2 with ESP32
Finding devices:
Finding devices:
Finding devices:
Finding devices:
```