adafruit/circuitpython #10059: Hard fault while accessing connection services.

iDevy, 2025-02-13:
Using circuitpython 9.2.4 on an M5stack Atom S3 lite.

installed libs:
- neopixel
- adafruit_requests
- adafruit_connection_manager
- bluepad32
- adafruit_esp32spi
- adafruit_bus_device
- adafruit_ble

I'm trying to connect to Lego WEDO brick to send commands (haven't got there yet).
Also first version of code (with a more generalized approach: `start_scan(Advertisement)`) worked for scanning, but returned too little information - no device names (usually use it to find the device to connect).

Current (problematic) code is as follows:

```python
import time
import board
from digitalio import DigitalInOut, Direction, Pull
import neopixel

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.uuid import UUID, VendorUUID

# Init button
btn = DigitalInOut(board.BTN)
btn.direction = Direction.INPUT
btn.pull = Pull.UP

# Init LED
led = neopixel.NeoPixel(board.NEOPIXEL, 1)
led.brightness = 0.1
values = (
    (255, 0, 0), 
    (128, 128, 0), 
    (0, 255, 0),  # Green
    (0, 128, 128),  # cyan
    (0, 0, 255),  # Blue
    (128, 0, 128), # Purple
)
INDEX = 0

# Init BLE
# Adapter = Adapter()
#_bleio.start_scan(extended=True, timeout=10, interval=1.2)

found = set()
scan_responses = set()
ble = BLERadio()
connection = None
print("scanning")

led[0] = (0,128,128)

WEDO_SERVICE_ID = VendorUUID('00001523-1212-efde-1523-785feabcd123')
WEDO_SERVICE = None

for advertisement in ble.start_scan(ProvideServicesAdvertisement):
    if advertisement.services:
        if WEDO_SERVICE_ID in advertisement.services:
            WEDO_SERVICE = list(advertisement.services)[0]
            led[0] = (0, 255, 0) 
            ble.stop_scan()
            connection = ble.connect(advertisement.address)
            connection.connection_interval = 11.5
            break
        
#t: UUID('00001523-1212-efde-1523-785feabcd123')> >
if connection:
    print(dir(connection)) 
    #connection.pair()  ## <-- Wedo doesn't seem to pair (it just hangs here indefinetely)
    print(connection.paired)
    # print('Service:', connection[WEDO_SERVICE])     ## <-- Uncommenting this line causes crash
    # print('Service:', connection[WEDO_SERVICE_ID]) ## <-- This one also crashes
    print('Con:\t\t', connection)
    print('List:\t\t', list(connection)) ## <-- This one returns AttributeError: 'int' object has no attribute 'bleio_uuid'
    print('Con dict: \t', connection.__dict__)
    #print('Con _dr: \t', connection._discover_remote(connection))  ## also crash
    #print('Con drs: \t', connection._bleio_connection.discover_remote_services()) ## and here too.
 
while True:
    if not btn.value:
        INDEX = 0 if INDEX > 4 else INDEX+1
    else:
        pass

    time.sleep(0.1) # sleep for debounce
    led[0] = values[INDEX]  # Red
    time.sleep(0.5)

```

The result in serial is:
```
Con:             <BLEConnection object at 0x3fcdca20>
Con dict:        {'_constructed_services': {}, '_bleio_connection': <Connection>, '_discovered_bleio_services': {}}
Traceback (most recent call last):
---- Closed serial port COM16 due to disconnection from the machine ----
---- Reopened serial port COM16 ----
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```

Cant even figure out where traceback tried to point.
Started to rework BLE part to pure `_bleio`, but I start suspecting that won't help.

tannewt, 2025-02-13:
I transferred this to the CircuitPython repo since it is a hard crash.

iDevy, 2025-03-23:
Is there any instrumentation to debug it futher or clarify the reasons?

tannewt, 2025-03-27:
You can build CircuitPython with `DEBUG=1` on the S3 to output the ESP-IDF logs to the UART. That's where I start usually.

dhalbert, 2025-08-25:
I actually have one of these WeDo 2.0 kits and can test this directly.