adafruit/circuitpython #10061: Remove original Biquad type from synthio

jepler, 2025-02-13:
In #9804 we deprecated the original synthio biquads in favor of the new BlockBiquads. For version 10, make the incompatible change of actually removing the old biquads.

dhalbert, 2025-03-28:
@relic-se I started to work on this, but I'm unfamiliar with the code. Re the struct `biquad_filter_state` and the associated routines in `synthio/Biquad.h`: should these be moved to BlockBiquad, or are they completely unnecessary? The are used in `audiofilters/Filter.c`, `synthio/Note.c` etc.

Also, is the `AnyBiquad` type now unneeded? Are they going to be a `BlockBiquad`? Should we consider renaming `BlockBiquad` to just `Biquad`, deprecating `BlockBiquad`, and removing the old name in CPy 11?

If you have a clear idea of the changes for this issue, and could do a PR, that would be great. Thanks.

relic-se, 2025-03-28:
I'll need to do some more digging, but I am definitely in favor of renaming `BlockBiquad` to `Biquad`. And you are correct on `AnyBiquad` which is just an amalgamation of the old and new classes.

I'd be happy to work on the PR here! I'll see if I can get something drafted this weekend.

closed 2025-04-01 by dhalbert