adafruit/circuitpython #10072: Improve how non-core socket implementations get their value of SOCK_STREAM.

jepler, 2025-02-16:
Recently in Discord, @tannewt encountered that zephyr uses different numeric values for some socket-related constants (vs lwip)

For core ssl support to work with pure Python socket implementations, such as wiznet, we had to promise that socket constants used by the ssl implementation would be the same numeric constants always, regardless of the internal implementation of sockets.

I think that the sole constant we depend on is is `SOCK_STREAM = 1`. This is because SSL can work on STREAM type sockets but not DGRAM type sockets.

@tannewt suggested that instead of the current approach of promising `SOCK_STREAM = 1` (the lwip value, used by raspberrypi & espressif), anytime we enable ssl in the core we could provide a minimal `socket` library with `SOCK_STREAM = (some value)`, not necessarily 1.

Then, wiznet would have to do something like
```
try:
    from socket import SOCK_STREAM
except ImportError:
    SOCK_STREAM = 1
SOCK_DGRAM = SOCK_STREAM + 1
```

wiznet currently has to adapt to different SOCK_ type numbering anyway, because wiznet uses 0x21 for SOCK_STREAM. (though it does so in a somewhat clever way that assumes the current values of 1 & 2)

Marking as a 10.0 milestone because undoing the promise that SOCK_STREAM=1 is an incompatible change.

mikeysklar, 2026-11-02:
@dhalbert and @jepler - 

Zephyr's `SOCK_STREAM` is also 1, the same as lwIP, so the mismatch that started this doesn't exist.

Changing the value would break ConnectionManager's check and the Wiznet5k mapping, and nothing needs it today.

Can we keep `SOCK_STREAM = 1` and close this?

jepler, 2026-11-02:
I don't have any further information than what I included in the initial comment, it's well out of my working memory several years later. I did find the discord discussion I alluded to in that comment but it hasn't illuminated much either: https://discord.com/channels/327254708534116352/327298996332658690/1338665356511674448

dhalbert, 2026-11-02:
Yes, let's close.

closed 2026-11-02 by dhalbert