adafruit/circuitpython #10074: How to "compile in" code for a product?

todbot, 2025-02-16:
CircuitPython is old enough now that people (including myself) are selling products using it. 
But installing the firmware for these products is pretty onerous for the end user: 
- Trigger UF2 bootloader
- Install CircuitPython
- Wait for CircuitPython to present CIRCUITPY drive
- Copy over needed libraries to CIRCUITPY
- Copy over `settings.toml` and `code.py`
- Reset
- Be sure to watch REPL in case you didn't copy the library files correctly!

In the case of RP2040/RP2350, we can at least use `picotool` to create a UF2 of a working system and distribute that. 

But is there a way to do this for the general case?  Those who make ESP32-based products would really benefit I think. 

As a stopgap, I created https://github.com/adafruit/circuitpython/pull/10073 to let us freeze-in some functionality, maybe a glorified "hello world" that shows the unit works.   But it would be really neat if I could tell the CircuitPython build system "here's a directory, make it the CIRCUITPY drive".  I realize this may need some help from the platform-specific UF2 bootloaders, but maybe do-able?

dhalbert, 2025-02-16:
esptool.py also has dump/load options that can do all of flash. So you could use it like you use picotool.

todbot, 2025-02-16:
Agreed, but requiring the user to install esptool.py for product updates is also asking a lot.

dhalbert, 2025-02-16:
> Agreed, but requiring the user to install esptool.py for product updates is also asking a lot.

You can point the user to https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ or similar browser tools. Especially on Espressif chips, since the UF2 bootloader is not protected, they may need to be familiar with it anyway.

tannewt, 2025-02-18:
> Agreed, but requiring the user to install esptool.py for product updates is also asking a lot.

Do we really need CircuitPython to support this? Would it be enough to have a folder -> fat filesystem -> uf2 converter? This would be orthogonal to CircuitPython except for placing the filesystem in the same place.

I don't really want to bundle with CP itself because it'll overwrite the user's file system.

tannewt, 2025-02-18:
Also, I think many other UF2 bootloaders besides the RP2 version have a `current.uf2` file that images the whole filesystem.

todbot, 2025-02-18:
> Do we really need CircuitPython to support this? Would it be enough to have a folder -> fat filesystem -> uf2 converter? This would be orthogonal to CircuitPython except for placing the filesystem in the same place.

This would be ideal. Is there a way to pull out of a CircuitPython UF2 all the information needed to append the CIRCUITPY filesystem?  It seemed to me this hypothetical tool would need to know some internals of particular CircuitPython builds in order to do that.  And/or would need a modified UF2 bootloader for those systems with internal+external flash. 
 
> I don't really want to bundle with CP itself because it'll overwrite the user's file system.

If we could do it outside of CirPy, that'd be awesome. The current stop-gap PR works just like the "make a default code.py when no filesystem exists" so there's no worries about filesystem overriding. But point taken: CircuitPython programmers are used to the filesystem staying intact when upgrading CircuitPython version. 

> Also, I think many other UF2 bootloaders besides the RP2 version have a current.uf2 file that images the whole flash including the filesystem.

afaik, only the SAMD21 bootloader can do this.

jepler, 2025-02-18:
I wrote this a long time ago, thinking it might be useful for shipping prepared CIRCUITPY filesystems, but it was never used and so I archived it: https://github.com/jepler/mkfatimg

tannewt, 2025-02-18:
> This would be ideal. Is there a way to pull out of a CircuitPython UF2 all the information needed to append the CIRCUITPY filesystem? It seemed to me this hypothetical tool would need to know some [...]

The filesystem isn't "appended" necessarily. It wouldn't be easy to get the location out of a UF2. You might be able to get it out of an elf with debug info. The location doesn't vary build to build because we don't want to destroy user file systems. The location may only change on major version boundary.

What platform are you targeting specifically? The implementation will depend on that.

The broadcom port builds its own SD card filesystem image here:
https://github.com/adafruit/circuitpython/blob/457edc304a96c64596ae423ca9e3eebd3641fa6d/ports/broadcom/Makefile#L150-L156

todbot, 2025-02-18:
> The filesystem isn't "appended" necessarily. 

Ah okay.  I phrased it that way because at some point I recall having success literally concatenating UF2 files to append the CIRCUITPY filesystem on rp2. Looks like there's things like [uf2tool](https://github.com/pguyot/uf2tool) now to be little less cavalier about it. 

> What platform are you targeting specifically? The implementation will depend on that.

For architectures, I think rp2 and esp32 would cover most systems people are concerned about. (The lack of something like `picotool` to produce full CirPy+filesystem for esp32 has definitely steered me away that using that chip in a product these last few years)

RetiredWizard, 2025-02-19:
The Blues.io circuitpython-fs-builder might be of interest to you, https://github.com/blues/circuitpython-fs-builder

I stumbled across it while playing with the Swan microcontroller a couple years ago. It's really meant for a different use case, but I thought it might be useful for the use case I think you're talking about here. I'm easily distracted though and didn't really follow up. There's a pretty good video explanation starting at around 50:20 of this video:  https://youtu.be/fUUOmpEwpCE

tannewt, 2025-02-19:
> For architectures, I think rp2 and esp32 would cover most systems people are concerned about.

rp2 should be possible already. The bootloader allows you to overwrite any portion of flash.

ESP32 *doesn't* allow a uf2 to write outside of the ota_0 partition. ([tinyuf2 README](https://github.com/adafruit/tinyuf2/tree/master/ports/espressif) So, we'd need to add the ability to write to the data partition.

Alternatively, idf.py could be used to flash directly. It can include both TinyUF2, CircuitPython and the filesystem.

bill88t, 2025-03-06:
Regarding https://github.com/adafruit/circuitpython/pull/10073:
I have an idea. A whole fs into an oneliner. I'm doing the generation script right now.
If I just disable auto-reload, import os, and blast things in `code.py`, eventually overwriting `code.py` itself, it's gonna work perfectly.

mikeysklar, 2026-09-01:
I wrote a small external tool along the lines you suggested: folder to FAT image to UF2 for rp2, https://github.com/mikeysklar/folder2uf2. Single stdlib-only Python file. It replicates the FAT geometry CircuitPython's own `f_mkfs` produces and writes only the CIRCUITPY region, so firmware and filesystem can be flashed independently, in either order.

Verified on a Metro RP2350 running 10.3.0-rc.0. Packed a folder with `code.py`, a `lib/` package and a 100KB binary, flashed the UF2, and the board booted straight into the shipped `code.py`. Every file checksummed byte for byte over the REPL. A 113 file, 1.8MB tree worked the same way.

| board | family | drive offset | fs size |
|---|---|---|---|
| raspberry_pi_pico | 0xe48bff56 | 0x100000 | 1.0MB |
| raspberry_pi_pico2 | 0xe48bff59 | 0x100000 | 3.1MB |
| raspberry_pi_pico_w | 0xe48bff56 | 0x181000 | 0.5MB |
| adafruit_metro_rp2350 | 0xe48bff59 | 0x100000 | 15.7MB |

Offsets are derived from `CIRCUITPY_FIRMWARE_SIZE` and `CIRCUITPY_CIRCUITPY_DRIVE_START_ADDR` in `ports/raspberrypi/mpconfigport.h`. Eight boards have built-in profiles and anything else works with `--flash-offset`, `--flash-size` and `--family-id`.

One thing worth knowing for anyone reading the raw flash: `supervisor/shared/flash.c` synthesizes an MBR at read time with partition 1 at LBA 1, so what is actually stored is a bare FAT volume with hidden-sectors set to 1. The tool matches that, and I checked it against the boot sector the board itself reports before writing anything.

ESP32 stays out of scope, since tinyuf2 only writes ota_0 as you noted above. Credit to jepler's mkfatimg for exploring this first.

Written with an LLM agent. I derived the offsets from the source and ran the hardware tests myself.

todbot, 2026-09-01:
Thanks @mikeysklar! I'd love it if we could get something similar for ESP32 but not sure how. For RP2040/RP2350, I've been using `picotool` to get an entire copy of the flash of a working board to use as a full download for customers.  On ESP32 can do a .bin and use esptool.py but UF2 would be so much cooler.

mikeysklar, 2026-09-02:
@todbot Added `--self-extract` for this. Bundle your product folder into one file:

```sh
folder2uf2 --self-extract -o code.py myproject/
# code.py: 4 files, 60144 bytes of source packed into 83952 bytes
```

Ship customers that single `code.py`. They drag it onto CIRCUITPY:

```
1. drag code.py onto CIRCUITPY
2. CIRCUITPY disappears for about 4 seconds while it unpacks
3. board reboots with code.py, lib/ and assets in place
```

The payload rides in trailing comments. `storage.unsafe_disable_usb_drive()` makes CIRCUITPY writable, then it inflates one file at a time.

No bootloader, no esptool, no reset button. Verified on a Metro RP2350 and a Metro ESP32-S3, both on 10.3.0.

https://github.com/mikeysklar/folder2uf2

tannewt, 2026-09-02:
> https://github.com/mikeysklar/folder2uf2

Please package this and put it on pypi so it is pip installable. Then we can close this.

mikeysklar, 2026-09-02:
@tannewt - Will do. Learn guide imminent.

mikeysklar, 2026-09-03:
@todbot - Let me know how this works out for you. 

It survived my standard, sub-standard testing.

On PyPI now:

```
pip install folder2uf2
```

https://github.com/mikeysklar/folder2uf2
https://pypi.org/project/folder2uf2/

mikeysklar, 2026-09-04:
Tested `--self-extract` across the ports on my HIL farm. Every row was run on hardware, backed up and restored afterwards.

| Chip | Tested on | Self-extract |
|---|---|---|
| RP2040 | Metro RP2040 | yes |
| RP2350 | Metro RP2350 | yes |
| ESP32-S2 | Metro ESP32-S2 | yes |
| ESP32-S3 | Metro ESP32-S3 | yes |
| SAMD51 | Metro M4 AirLift | yes |
| nRF52840 | Feather nRF52840 | yes |
| STM32F405 | Feather STM32F405 | yes |
| SAMD21 | Metro M0 Express | no |

SAMD21 is not going to happen. The build has no `zlib` and no `binascii`, so there is nothing to unpack with.

todbot, 2026-09-07:
Thanks @mikeysklar!  I do wish we could do full-image UF2s for ESP32S* but I suppose that would be a job for tinyuf2, if it's even possible. Now that Firefox has WebSerial, it's a bit easier for normies to flash .bins. I'll close this.

closed 2026-09-07 by todbot