adafruit/circuitpython #10075: Will there be support for pulseio to be added to the Trinket M0?

7ator, 2025-02-16:
I purchased the Adafruit Trinket M0, not realizing it didn't have support for `pulseio`, my mistake, I should've checked. 
I was just wondering if there will be any possibility for current or future implementation of the `pulseio` module for the Trinket M0?

dhalbert, 2025-02-16:
There is an implementation but it doesn't fit in the 192kB of firmware space available on the Trinket M0. It's on other SAMD21 boards and also many other more capable boards of similar size, such the as QT Py RP2040.

7ator, 2025-02-16:
> There is an implementation but it doesn't fit in the 192kB of firmware space available on the Trinket M0. It's on other SAMD21 boards and also many other more capable boards of similar size, such the as QT Py RP2040.

Gotcha, appreciate the response!

closed 2025-02-16 by 7ator

dhalbert, 2025-02-16:
Check https://docs.circuitpython.org/en/latest/shared-bindings/support_matrix.html and type `pulseio` in the search box to see which boards support it.