adafruit/circuitpython #10078: Countio for ItsyBitsy M0 Express?

rp10007, 2025-02-18:
I can see why it wouldn't fit in builtins (although rotaryio made the cut), but is there a way to have it as a loadable mpy that could fit on local storage? Is there code that I could cross-compile, or is that not for ordinary mortals?

thanks!

dhalbert, 2025-02-18:
It is implemented in C and is not loadable at run-time. MicroPython has some provisions for that kind of thing, but we don't (and that mechanism takes space as well).

You could make a custom build for your board: https://learn.adafruit.com/building-circuitpython/customizing-included-modules (but read the whole guide).

rp10007, 2025-02-19:
Thanks! I think I'll swap in a different board for now.

closed 2025-02-19 by rp10007