adafruit/circuitpython #10096: CP drive does not appear with Partilce Boron

cadet702, 2025-02-24:
Adafruit CircuitPython 9.2.4 on 2025-01-28; Particle Boron with nRF52840.

I'm struggling to get a Particle Boron (https://www.adafruit.com/product/3994) to display the CircuitPython drive.  In case it's helpful, the version printed on the PCB is v1.5.0, the model number on the sticker is BRN404X, FCC ID is 2AEMI-BRN404X, and the IC is 20127-BRN404X.

I used a J-Link Edu to replace the stock bootloader with the Boron's UF2 bootloader, but when I copied the .uf2 file for CP to the Boron, no CircuitPython drive appeared.  This appears to be a Boron-specific thing because, I was able to perform the same sequence of operation on the Partilce Argon (with Argon appropriate files of course) and did get the CircuitPython drive to appear.

I checked the common issues and I don't have any of the offending programs like anti-virus software or Samsung Magician that are known to cause this issue.  I also found an old forum post where someone said they had to uninstall the device drivers in windows to get it to show up as a drive-type-device, and I tried that, but no joy.

In case it's helpful I've tried CP 9.2.4 with the following bootloaders:
Boron (I also tried CP 9.2.4, 9.2.2, 8.2.9, 7.2.0, and 6.3.0 on this one)
Argon
Xenon
Generic NRF52480

All showed up as USB drives until I copied over the .uf2 file for CP.

Based on the blinking pattern of the RGB LED (3 white blinks every once in a while), I have inferred that CP appears to be running in safe mode, and was able to connect to the REPL using Putty.  I tried sending `import storage` followed by `storage.erase_filesystem()`, but this just caused the device to crash and reboot.  I've attached a screen shot showing that the boot loader appears as a drive and the contents of the REPL.

[image]

If it helps with diagnosis, I'd be happy to try loading a version of CP that doesn't attempt to use external flash, but I don't have the skills to compile this myself.

dhalbert, 2025-02-24:
If you connect to the REPL via a terminal program, and press reset, does it say you are in safe mode because it couldn't create CIRCUITPY?

This could be because the external flash on the board is not an expected part number. What does the printing on the external flash chip say?

cadet702, 2025-02-24:
There was one chip with markings visible that might be the flash.  The complete set of markings are:
`mx25 r6435f 8j606 000-x`

When I googled this, I noticed there were a couple links to forum posts related to issues using this flash in other devices (STM32U5 and nRF9160).

cadet702, 2025-02-25:
Success!  Boot_out.txt reads:
`Adafruit CircuitPython 9.2.4-194-ged3b5087e2 on 2025-02-24; Particle Boron with nRF52840
Board ID:particle_boron
UID:BC5C7AA50C685D7C`

dhalbert, 2025-02-25:
OK! We'll merge it and it will be in 9.2.5.

PR #10098 (2025-02-25): nordic/boards/particle_boron: add 8MB flash chip
- Fixes #10096 

Add another flash chip for Particle Boron. User's sample has an 8MB chip.
closed 2025-02-25 by dhalbert