adafruit/circuitpython #10120: Feather rp2040 with USB Type A - crashes when connecting keyboards

scrambler803, 2025-03-06:
## Description  
I'm trying to use a USB keyboard with the **Feather RP2040 with USB Type A** in CircuitPython (version **9.2.4**). However, when I plug in my keyboards, the Feather **crashes and disconnects from the PC after about 15 seconds**.  

I tested using [Adafruit’s USB Host device info example](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/usb-host-device-info-2), and the behavior is consistent. **A wired USB mouse works fine and is detected correctly, but both keyboards I tested cause the Feather to crash.**  

## Keyboards Tested  
1. **Apple USB Keyboard** (with built-in USB hub and media keys)  
2. **Custom keyboard running QMK firmware on a Raspberry Pi Pico**  

## Troubleshooting Steps Taken  
- Plugging in a **wired USB mouse** works fine, with proper enumeration output.  
- Tested with **KMK firmware** instead of QMK—same crash.  
- Verified the **Feather RP2040 board is powered** (green LED on).  
- Attempted **listing USB devices** using a modified `device_info.py` script but the Feather crashes upon keyboard connection.  
- Tried **flashing an experimental firmware build** provided in Discord—still crashes.  
- Suspected **composite USB devices (e.g., keyboards with extra media keys) might be the issue**, but not confirmed.  
- A developer (**tannewt**) was able to **reproduce the crash with a Mac keyboard**.  
- ### Additional Troubleshooting with QMK Firmware
  - To rule out compatibility issues, I attempted modifying the QMK firmware on my custom keyboard to ensure it enumerates as a single, basic HID device rather than a composite device.

  - Changes attempted in rules.mk:
    ```
    make
    Copy
    Edit
    NKRO_ENABLE = no
    EXTRAKEY_ENABLE = no
    CONSOLE_ENABLE = no
    MOUSEKEY_ENABLE = no
    KEYBOARD_SHARED_EP = no  # Previously set to "yes"
    ```
  - Despite these changes, the keyboard still causes the Feather to crash when connected.

  - #### Key insight from QMK developers:

    - Setting `KEYBOARD_SHARED_EP = no` should have forced the keyboard to enumerate as a basic 6KRO HID device.
    - Even after doing this, the Feather still crashes upon connection.
    - This suggests the issue may not be related to composite vs. single USB device enumeration but rather a deeper incompatibility between certain HID keyboards and the Feather’s USB host implementation.

## Expected Behavior  
The Feather should enumerate the keyboard as a USB device and allow input handling, similar to how it correctly detects a mouse.  

## Actual Behavior  
- Upon connecting a keyboard, the Feather **freezes for about 15 seconds**, then **crashes and disconnects** from the PC.  
- Issue occurs **even without running code.py**—the Feather crashes just from plugging in the keyboard.  

## CircuitPython Version  
- **Firmware:** `adafruit-circuitpython-adafruit_feather_rp2040_usb_host-en_US-9.2.4.uf2`  

## Possible Causes?  
- Incompatibility with **composite USB devices** (keyboards with multiple endpoints for media keys, hubs, etc.).  
- TinyUSB limitations in handling **certain keyboard USB descriptors**.  
- Insufficient **power or USB negotiation issues** with the Feather RP2040.  

## Additional Context  
**tannewt** suggested that USB enumeration failures might be causing the crash, and **Thach (who works on TinyUSB) might be able to look into it.**  

Would appreciate any insights on **what types of keyboards are supported** and whether this is a **known issue with TinyUSB on the Feather RP2040**.

tannewt, 2025-03-06:
I was able to reproduce this on the USB host feather wing *but* not on the RP2040 feather with a usb host breakout cable plugged in. The breakout cable's power pin was connected to the `USB` power pin on the feather header.

I suspect a power issue because the crash doesn't result in a safe mode. The device disappears and doesn't come back. I mean to check my Linux `dmesg` output but have run out of time. I'd also check the USB activity with an analyzer to see if anything happens before the crash. If nothing does, then power is a likely cause.

dhalbert, 2025-03-17:
Retry on Fruitjam with Apple keyboard.