adafruit/circuitpython #10125: Implement DMA-capable Memory Allocation for RP2350

eightycc, 2025-03-10:
For RP2350 boards with PSRAM, buffers for use with DMA may be unpredictably allocated in PSRAM. When DMA attempts to access a buffer in PSRAM, a bus fault occurs. This will manifest as a `CircuitPython core code crash`.

In the present implementation, SRAM and PSRAM are added to the main `tlsf` heap as separate pools. `tlsf` does not differentiate between these pools. It happens that `tlsf` will allocate memory from the SRAM pool first because it was added first. Once SRAM is filled, allocations will come from the PSRAM pool. As the `tlsf` heap ages, it becomes unpredictable from which pool allocations will come.

The `dma_capable` argument of `port_malloc()` is ignored by its RP2 port specialization. Likewise, `tlsf` does not provide a means of specifying which pool, SRAM or PSRAM, to use for an allocation.

Similarly, it is not predictable whether a VM heap space allocation is in SRAM or PSRAM. There is no means of requesting a VM heap space allocation in SRAM.

An example solution is to implement the `dma_capable` argument of `port_malloc` and allocate all DMA buffers from the main `tlsf` heap rather than the VM heap.

A workaround may be to allocate all DMA buffers early in the life of the system.

tannewt, 2025-03-11:
I think we'll need to improve common-hal memory use for this too then. I think we assume that VM allocated memory can be DMA'd.

jepler, 2025-03-11:
I think the low level Espressif APIs mostly bounce everything through SRAM that might be DMA'd, so it hasn't caused problems that e.g., a bytes argument to be sent out SPI is in PSRAM...

tannewt, 2025-05-12:
Done in #10240

closed 2025-05-12 by tannewt