adafruit/circuitpython #10133: usb_cdc module AttributeError on Adafruit KB2040

martinsram3k, 2025-03-17:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4 on 2025-01-28; Adafruit KB2040 with rp2040
```

### Code/REPL

```python
import usb_cdc
import board

usb_serial = usb_cdc.serial

while True:
    if usb_serial.in_waiting > 0:
        data = usb_serial.readline().decode('utf-8').strip()
        print(f"Přijatá data: {data}")
        usb_serial.write(f"Odpověď: {data}\r\n".encode())
```

### Behavior

**Actual Behavior:**

When attempting to use `usb_cdc.serial` or `usb_cdc.serials`, I receive the error `AttributeError: 'module' object has no attribute 'serial'` or `'module' object has no attribute 'serials'`.

**System Information:**

* Board: Adafruit KB2040
* CircuitPython Version: 9.2.4
* Computer Operating System: [Windows 10]
* my board si flashing red

### Description

Problem Description:

On my Adafruit KB2040 running CircuitPython 9.2.4, I am unable to initialize the serial port using the usb_cdc module. When attempting to use usb_cdc.serial or usb_cdc.serials, I receive the error AttributeError: 'module' object has no attribute 'serial' or 'module' object has no attribute 'serials'.

Steps I have taken:

Installed CircuitPython 9.2.4 on my Adafruit KB2040.
Verified the compatibility of the board and CircuitPython version.
Reinstalled CircuitPython multiple times.
Verified basic board functionality with a blinking LED test.
Performed a "factory reset" of the KB2040.
Checked the boot_out.txt file (attached below).
Tried different serial terminals and computers.
Verified that I have the USB-to-UART adapter connected correctly.
Tested bidirectional communication using PuTTY and code in code.py.
Expected Behavior:

I expect the usb_cdc module to function and allow me to communicate with the KB2040 via the serial port.

Neradoc, 2025-03-17:
Have you perused the documentation for the `usb_cdc` module ?
There is no such thing as `usb_cdc.serial` or `usb_cdc.serials`.
https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#module-usb_cdc

Here, I have some sample code for communicating via USB serial:
https://github.com/NeraSamples/circuitpython_usb_serial

dhalbert, 2025-03-25:
Closing because I think this has been addressed.

closed 2025-03-25 by dhalbert