adafruit/circuitpython #10140: Hard fault reading input from keyboard

FoamyGuy, 2025-03-18:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4-274-g2af397ed1c on 2025-03-06; Adafruit Metro RP2350 with rp2350b
Board ID:adafruit_metro_rp2350
```

### Code/REPL

```python
import supervisor
import sys

while True:
    available = supervisor.runtime.serial_bytes_available
    if available:
        c = sys.stdin.read(available)
        print(c, end='')
```

### Behavior

The code runs and waits until a key is pressed. Once a key is pressed the device reboots and there is this hardfault message in the serial console:
```
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

### Additional information

Using this keyboard: https://www.adafruit.com/product/857

FoamyGuy, 2025-03-18:
There seems to be something specific about this keyboard that it doesn't like. I've tried the following and they all work with the same code on the same device:

- IOGear GKM561R https://iogear.com/products/gkm561r
- Logitech K400 https://www.logitech.com/en-us/shop/p/k400-plus-touchpad-keyboard
- Kinesis Advantage https://kinesis-ergo.com/support/advantage/

tannewt, 2025-05-01:
This still happen?

FoamyGuy, 2025-05-02:
`9.2.7` does still hard fault when keys are pressed sometimes. `10.0.0-alpha.2` seems like it will not hard fault, I have tested it and pressed away on the keyboard for a while and never got one. With `9.2.7`, and originally in `9.2.4` when this was submitted it occurs fairly fast within dozen or so key presses. 

None of the versions successfully read the keys and print them on Metro RP2350 connected to USB host pins, but that is less problematic than the hard fault.

hathach, 2025-05-05:
sure thing, will get it next order