adafruit/circuitpython #10155: RP2040-GEEK - supervisor USB identification cannot be set

HidingCherry, 2025-03-20:
### CircuitPython version and board name

```python
CircuitPython 9.2.5 (also tried 9.2.4); Waveshare RP2040-GEEK
```

### Code/REPL

```python
# boot.py
import board
import storage
import usb_cdc

import supervisor
supervisor.set_usb_identification(
	vid=0x0123,
	pid=0x0123,
	manufacturer="manu",
	product="prod"
)
```

### Behavior

No error.

Device should be identified as:

usb 1-4: New USB device found, idVendor=0123, idProduct=0123, bcdDevice= 1.00
usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-4: Product: prod
usb 1-4: Manufacturer: manu

Instead it is identified as this:

usb 1-4: New USB device found, idVendor=2e8a, idProduct=1056, bcdDevice= 1.00
usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-4: Product: RP2040-GEEK
usb 1-4: Manufacturer: Waveshare Electronics

### Description

On boot the usb description should be set accordingly.

### Additional information

This issue happens with the rp2040_geek images, but it does not happen with rp2040_pico (which I falsely downloaded first).

HidingCherry, 2025-03-25:
ArchLinux - up2date zen kernel

todbot, 2025-04-14:
@HidingCherry, are you doing a full reset or power cycle of the RP2040-GEEK board? The `boot.py` file is only run on hardware reset. 
Also, most OSes cache USB descriptors, so some changes like manufacturer & product strings, will not be updated in `lsusb`. 
Whenever I change `supervisor.set_usb_identification()` settings, I plug it into (another) computer that's been freshly power-cycled.  This guarantees I'm not hitting any OS USB descriptor cache and that `boot.py` is being run.  
btw, there's nothing different in the USB config for `wavewhare_rp2040_geek` compared to `raspberry_pi_pico`.  Any issues you're experiencing would happen with either CircuitPython build

HidingCherry, 2025-04-14:
I unplug and replug it on the same port.
I have the same code running on both builds.
Still the waveshare_rp2040_geek gives me the wrong descriptor after several tests, but raspberry_pi_pico works without issues.

I can test with different ports later this week, but I'm pretty sure the results will be the same.
I will test with current firmware ofc

HidingCherry, 2025-05-01:
It's a user error, sorry for the trouble.
I had a few more lines - it seems that if a line gives an error, all lines behind those won't run.

The lines were for button handling, which were set wrong.
I think the GPIOs which were set by me are used by the display, as such those lines error out with the result that my supervisor settings did not work.

I'll close this issue then.

closed 2025-05-01 by HidingCherry