adafruit/circuitpython #10161: Problems with accessing SD CARD on  ESP32-S3-GEEK board

donkiejkong, 2025-03-22:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-S3-GEEK with ESP32S3
```

### Code/REPL

```python
import os
import board
import sdioio
import storage
import bitbangio

CMD_PIN = 35
CLK_PIN = 36
D0_PIN  = 37
SD_SLOT_WIDTH = 1

# example 1
spi = bitbangio.SPI(board.GP36, board.GP35, board.GP37) # always fails
cs = digitalio.DigitalInOut(board.DAT3)
sd = adafruit_sdcard.SDCard(spi, cs) 

# example 2 (always fails)
sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37], frequency=25000000)

# example 3 (alway fails)
spi = busio.SPI(board.SD_SCK, MOSI=board.SD_MOSI, MISO=board.SD_MISO) 

# the usual things
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')
os.listdir('/sd')
```

### Behavior

the error is always the same:
Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-S3-GEEK with ESP32S3
Board ID:waveshare_esp32_s3_geek
UID:C34872620D8B
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 15, in <module>
ValueError: GP36 in use   # always same error for all of the examples

### Description

Hello,

This is my device:
https://www.waveshare.com/wiki/ESP32-S3-GEEK

I've installed python circuit from here: https://circuitpython.org/board/waveshare_esp32_s3_geek/

I'm trying to port one of the examples from C to python:
https://github.com/user-attachments/files/19285971/waveshare_usb_msc_wireless_disk_example.zip

However I'm currently stuck trying to make the sd card working, this is my boot.py (i've tried multiple approaches):

```

import os
import board
import sdioio
import storage
import bitbangio

CMD_PIN = 35
CLK_PIN = 36
D0_PIN  = 37
SD_SLOT_WIDTH = 1

# example 1
spi = bitbangio.SPI(board.GP36, board.GP35, board.GP37) # always fails
cs = digitalio.DigitalInOut(board.DAT3)
sd = adafruit_sdcard.SDCard(spi, cs) 

# example 2 (always fails)
sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37], frequency=25000000)

# example 3 (alway fails)
spi = busio.SPI(board.SD_SCK, MOSI=board.SD_MOSI, MISO=board.SD_MISO) 

# the usual things
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')
os.listdir('/sd')

```
the error is always the same:
Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-S3-GEEK with ESP32S3
Board ID:waveshare_esp32_s3_geek
UID:C34872620D8B
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 15, in <module>
ValueError: GP36 in use

I want to use the SD card and make it visible to the operating system as mass storage device. (it works in the example in C). Anyway am i missing something here? why this error happens?

snkYmkrct, 2025-03-22:
There is a default SPI object defined for this board on GPIO35 to 37
https://github.com/adafruit/circuitpython/blob/a0b482c2806ac19962ed26f2d4bbe5ebdef3406a/ports/espressif/boards/waveshare_esp32_s3_geek/pins.c#L67-L69

If you want to use the sd card over SPI, with the `sdcardio` module, calling it like this `sd = sdcardio.SDCard(board.SPI(), board.SD_CS)` should work. 

Based on the board schematic, the SDIO interface is available too, so `sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37, board.GP33, board.GP38, board.GP34], frequency=25000000)` should work as well, if you don't try to use the SPI at the same time. 

For help with CircuitPython questions like these, it's better to ask in the  [Adafruit Discord server](https://adafru.it/discord)

Neradoc, 2025-03-22:
Hi, I looked at the board definition and there seems to be mistakes in there: it instantiates the SPI bus of the SD card (also defined as board.SPI) to use the display, despite the display being on a different SPI bus according to pins.c.
I don't have the board to test, can you check the following ?

- 1. Does the display show anything when the board is running ?
- 2. Try this code:

```py
spi = board.SD_SPI()
cs = digitalio.DigitalInOut(board.SD_CS)
sd = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')
os.listdir('/sd')
```

As for your ultimate goal of showing the SD card as mass storage, that's not gonna work over USB, it's not currently supported in Circuitpython.
For using it over wifi, you could use the web workflow for that, or write a server in python.
We have adafruit_httpserver, but I don't think that a file server has been done yet.

donkiejkong, 2025-03-22:
hello,

i've tried the code the error now is:

`Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-S3-GEEK with ESP32S3
Board ID:waveshare_esp32_s3_geek
UID:C34872620D8B
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 19, in <module>
  File "adafruit_sdcard.py", line 112, in __init__
  File "adafruit_sdcard.py", line 130, in _init_card
  File "adafruit_sdcard.py", line 252, in _cmd
  File "adafruit_sdcard.py", line 205, in _wait_for_ready
ValueError: No miso pin
`

line 19 of boot.py is "sd = adafruit_sdcard.SDCard(spi, cs)"

i'm using adafruit_sdcard.mpy lib from adafruit-circuitpython-sd-9.x-mpy-3.3.25

donkiejkong, 2025-03-22:
> There is a default SPI object defined for this board on GPIO35 to 37
> 
> [circuitpython/ports/espressif/boards/waveshare_esp32_s3_geek/pins.c](https://github.com/adafruit/circuitpython/blob/a0b482c2806ac19962ed26f2d4bbe5ebdef3406a/ports/espressif/boards/waveshare_esp32_s3_geek/pins.c#L67-L69)
> 
> Lines 67 to 69 in [a0b482c](/adafruit/circuitpython/commit/a0b482c2806ac19962ed26f2d4bbe5ebdef3406a)
> 
>  { MP_ROM_QSTR(MP_QSTR_SD_SCK),  MP_ROM_PTR(&pin_GPIO36)}, 
>  { MP_ROM_QSTR(MP_QSTR_SD_MOSI), MP_ROM_PTR(&pin_GPIO35)}, 
>  { MP_ROM_QSTR(MP_QSTR_SD_MISO), MP_ROM_PTR(&pin_GPIO37)}, 
> If you want to use the sd card over SPI, with the `sdcardio` module, calling it like this `sd = sdcardio.SDCard(board.SPI(), board.SD_CS)` should work.
> 
> Based on the board schematic, the SDIO interface is available too, so `sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37, board.GP33, board.GP38, board.GP34], frequency=25000000)` should work as well, if you don't try to use the SPI at the same time.
> 
> For help with CircuitPython questions like these, it's better to ask in the [Adafruit Discord server](https://adafru.it/discord)

thanks for the reply, i've modified the code to:

`
sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37, board.GP33, board.GP38, board.GP34], frequency=25000000
`

the error is still the same:
Traceback (most recent call last):
  File "boot.py", line 16, in <module>
ValueError: GP36 in use

Neradoc, 2025-03-22:
> `Adafruit CircuitPython 9.2.4 on 2025-01-29; Waveshare ESP32-S3-GEEK with ESP32S3 Board ID:waveshare_esp32_s3_geek UID:C34872620D8B boot.py output: Traceback (most recent call last): File "boot.py", [...]

Oh yeah that would actually not work...
The issue with the board definition is here:
https://github.com/adafruit/circuitpython/blob/a0b482c2806ac19962ed26f2d4bbe5ebdef3406a/ports/espressif/boards/waveshare_esp32_s3_geek/board.c#L34-L42

- It creates board.SD_SPI() making the SD pins occupied.
- It overwrites the spi variable with the LCD pins.

Therefore board.SD_SPI can't be used for the SD card, unlike I first thought, since it doesn't have the right pins anymore, despite the pins being busy. (And it doesn't have a MISO pin anymore). Plus the SD pins are unreachable at this point, you can't even deinit the SPI bus on them.

I believe I have a fix for that: #10162 can you test it from the artifacts once it's built ?

Neradoc, 2025-03-22:
[image]

RetiredWizard, 2025-03-22:
I'll also give the updated PR a test but for what it's worth the SDIO interface does seem to work on 9.2.5
```py
import sdioio
import board
import storage
sd = sdioio.SDCard(clock=board.GP36,command=board.GP35,data=[board.GP37, board.GP33, board.GP38, board.GP34],frequency=25000000)
vfs=storage.VfsFat(sd)
storage.mount(vfs,'/sd')
```

RetiredWizard, 2025-03-22:
> * Try this code:
> 
> 
> spi = board.SD_SPI()
> cs = digitalio.DigitalInOut(board.SD_CS)
> sd = adafruit_sdcard.SDCard(spi, cs)
> vfs = storage.VfsFat(sd)
> storage.mount(vfs, '/sd')
> os.listdir('/sd')

This works with the artifact.....

donkiejkong, 2025-03-22:
> FYI this is where it is: <img alt="Image" width="1163" src="https://private-user-images.githubusercontent.com/6160205/425747201-39e96172-5874-4851-baec-6c840eb739af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI2NTA1OTQsIm5iZiI6MTc0MjY1MDI5NCwicGF0aCI6Ii82MTYwMjA1LzQyNTc0NzIwMS0zOWU5NjE3Mi01ODc0LTQ4NTEtYmFlYy02Yzg0MGViNzM5YWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMjJUMTMzMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGRlMTRkNWZiMTU5NTE0OGQzZjM5ZWVjNDhkNzU2NzhjOWM2MTZhM2NkMTUxODM2MThhMzcxNjdhZTRjOTlhNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.C0LTa7DLuvdfDlViIvnbM6jy8dJgGmeHLz2AYJxwwPM">

thanks a lot this seems to work, at least i can now access the SD card!

any ideas how to mount the /sd to be visible by the operating system like a normal mass storage / USB thumb drive? (this works with flash memory "/", but not "/sd or "sd")

```
import os
import board
import sdioio
import storage
import bitbangio
import digitalio
import adafruit_sdcard
import supervisor

sd = sdioio.SDCard(clock=board.GP36, command=board.GP35, data=[board.GP37, board.GP33, board.GP38, board.GP34], frequency=25000000)
  
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')

print("### HELLO ###")
print(os.listdir('/sd'))

storage.remount('/sd', disable_concurrent_write_protection=True)
supervisor.runtime.autoreload = False
```

this line "storage.remount('/sd', disable_concurrent_write_protection=True)" fails with "OSError: [Errno 22] Invalid argument".

RetiredWizard, 2025-03-22:
The SDIO mount now doesn't work though...

```
>>> sd = sdioio.SDCard(clock=board.GP36,command=board.GP35,data=[board.GP37, board.GP33, board.GP38, board.GP34],frequency=25000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: SDIO Init Error 107
```

Neradoc, 2025-03-22:
> this line "storage.remount('/sd', disable_concurrent_write_protection=True)" fails with "OSError: [Errno 22] Invalid argument".

Yes at present time remount() only works on "/". 
As I mentioned above the SD card is not mounted and visible via USB MSC, only from python code.
There's a PR for that though: #10129, it will probably make its way in Circuitpython 10.

dhalbert, 2025-03-22:
This should be fixed by #10164

PR #10164 (2025-03-22): waveshare_esp32_s3_geek: fix SPI busses and board_init()
Adds board.SPI for the display SPI bus, and use it in `board.c`.

Previously the code accidentally instantiated both the SD bus (board.SD_SPI) and overwrote it with the display bus in display_init(), making the SD pins unusable. I can't test it.
Should fix #10161

(fixed rebase shenanigans... I think)
closed 2025-03-22 by dhalbert