adafruit/circuitpython #10167: Starting with PR10122 build, S3 boards can't storage.remount False in boot.py

RetiredWizard, 2025-03-22:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.4-310-gca0eec3f1a on 2025-03-11; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
```

### Code/REPL

```python
# Actually boot.py, not code.py
import storage
storage.remount("/",False)
```

### Behavior

```
Adafruit CircuitPython 9.2.4-310-gca0eec3f1a on 2025-03-11; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
Board ID:adafruit_feather_esp32s3_reverse_tft
UID:0740D1DC2800
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 2, in <module>
RuntimeError: Cannot remount '/' when visible via USB.
```

### Description

File system access can't be configured for microcontroller access starting with the `adafruit-circuitpython-adafruit_feather_esp32s3_reverse_tft-en_US-20250311-main-PR10122-ca0eec3.uf2` build.

### Additional information

I first noticed this on the Waveshare ESP32S3 Geek board and then tested the reverse TFT to make sure it wasn't a problem with the device.

RetiredWizard, 2025-03-22:
This is also happening on an Adafruit_feather_rp2350 so it's not limited to Esspressif/S3.....

dhalbert, 2025-03-22:
This will be a high priority to fix.

dhalbert, 2025-03-22:
Fixed by #10168

PR #10168 (2025-03-22): usb_msc_flash.c: restore initialization of `ejected` array
- Fixes #10167.

In #10122, the initialization of `ejected[]` was accidentally dropped in `supervisor/shared/usb/usb_msc_flash.c`.

Fix tested on a CPX.
closed 2025-03-22 by dhalbert