adafruit/circuitpython #10171: AttributeError: 'module' object has no attribute 'SHT31D'

iz8mbw, 2025-03-23:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.6 on 2025-03-23; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
import time
import board
import busio
import adafruit_sht31d
```

### Behavior

Adafruit CircuitPython 9.2.6 on 2025-03-23; Waveshare RP2040-Zero with rp2040
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  **File "code.py", line 15, in <module>
AttributeError: 'module' object has no attribute 'SHT31D'**

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

### Description

Hello.
Trying to read some I2C sensors with RP2040-Zero but without success.
I always recieve the error:  `AttributeError: 'module' object has no attribute`
In this case since using SHT31, the error is: `AttributeError: 'module' object has no attribute 'SHT31D'`

On the RP2040 "lib" I have these library:

[image]

This is the `code.py`:

```

# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import busio
import adafruit_bitbangio
import adafruit_sht31d

# Create sensor object, communicating over the board's default I2C bus
i2c = busio.I2C(board.GP29, board.GP28)
# i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller
sensor = adafruit_sht31d.SHT31D(i2c)
print("\033[1mSensor\033[0m = SHT31-D")
print("\033[1mSerial Number\033[0m = ", sensor.serial_number, "\n")
sensor.frequency = adafruit_sht31d.FREQUENCY_1
sensor.mode = adafruit_sht31d.MODE_PERIODIC
for i in range(3):
    print("Please wait...", end="\r")
    if i == 2:
        sensor.heater = True
    if i == 1:
        time.sleep(4)
        print("\033[91mCache half full.\033[0m")
    else:
        time.sleep(8)
    if sensor.heater:
        print("\033[1mHeater:\033[0m On    ")
        sensor.heater = False
    print("\033[1mTemperature:\033[0m ", sensor.temperature)
    if not sensor.heater:
        print("\033[1mHeater:\033[0m Off")
    print("\033[1mHumidity:\033[0m ", sensor.relative_humidity, "\n")
sensor.mode = adafruit_sht31d.MODE_SINGLE
```

Where is the issue?

Mnay thanks!

dhalbert, 2025-03-23:
I am not seeing this problem. I tested on a QT Py RP2040, with both CircuitPython 9.2.3 and 9.2.6. My test program is simpler:

```py
import time
import board
import busio
import adafruit_sht31d

#i2c = busio.I2C(board.GP29, board.GP28)
i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller
sensor = adafruit_sht31d.SHT31D(i2c)

while True:
    print(sensor.temperature)
    time.sleep(1)
```
I am suspicious that there may be some corruption of the .mpy files. Also, is there an `adafruit_sht31d.mpy` or `.py` in the top level?
Here is the size I expect with the latest `adafruit_sht31d.mpy`:
```
$ ls -l /media/halbert/CIRCUITPY/lib/adafruit_sht31d.mpy
-rw-r--r-- 1 halbert halbert 3681 Mar 23 09:32 /media/halbert/CIRCUITPY/lib/adafruit_sht31d.mpy
```

iz8mbw, 2025-03-23:
Yes!
The issue was the file `adafruit_sht31d.mpy`, no ideas why but the size was different.

Mnay thanks and sorry!

closed 2025-03-23 by iz8mbw