adafruit/circuitpython #10172: same54_xplained fails to boot when built with GCC 14

EmergReanimator, 2025-03-23:
### CircuitPython version and board name

```python
- CPY >= 9.0.0
- same54_xplained
```

### Code/REPL

```python
Not applicable
```

### Behavior

Not applicable

### Description

CPY stucks in file system initialisation when built with GCC14.
The same code works fine if it is compiled with GCC10;

Tried the tag 9.0.0;

Here is the back trace of problematic code. Currently no idea what exactly goes wrong here.
`
#0  0x00060e98 in memcpy (dst=dst@entry=0x2003fb1c, src=src@entry=0x4000000, n=n@entry=1) at ../../shared/libc/string0.c:52
#1  0x0004b748 in spi_flash_read_command (command=<optimized out>, command@entry=5 '\005', response=response@entry=0x2003fb1c "\002", length=length@entry=1) at supervisor/qspi_flash.c:77
#2  0x00055822 in wait_for_flash_ready () at ../../supervisor/shared/external_flash/external_flash.c:73
#3  0x00055a5c in write_flash (address=0, data=0x20002fc4 "\353\376\220MSDOS5.0", data_length=data_length@entry=256) at ../../supervisor/shared/external_flash/external_flash.c:120
#4  0x0004b6c2 in flush_ram_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:443
#5  spi_flash_flush_keep_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:465
#6  0x0004e5b6 in supervisor_external_flash_flush () at ../../supervisor/shared/external_flash/external_flash.c:474
#7  supervisor_flash_flush () at ../../supervisor/shared/flash.c:141
#8  0x00027f30 in external_flash_write_block (data=<optimized out>, block=8) at ../../supervisor/shared/external_flash/external_flash.c:541
#9  supervisor_flash_write_blocks (src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/external_flash/external_flash.c:577
#10 flash_write_blocks (self=<optimized out>, src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/flash.c:133
#11 flash_write_blocks (self=<optimized out>, src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/flash.c:120
#12 0x00022d7a in mp_vfs_blockdev_write (self=<optimized out>, block_num=9, num_blocks=1, buf=<optimized out>) at ../../extmod/vfs_blockdev.c:117
#13 disk_write (pdrv=<optimized out>, buff=<optimized out>, sector=9, count=1) at ../../extmod/vfs_fat_diskio.c:86
#14 0x00026772 in f_mkfs.constprop.0 (fs=<optimized out>, opt=opt@entry=7 '\a', work=work@entry=0x2003fd28, len=512, au=<optimized out>) at ../../lib/oofatfs/ff.c:5837
#15 0x000272fe in filesystem_init (create_allowed=true, force_create=false) at ../../supervisor/shared/filesystem.c:112
#16 0x0004bb4e in main () at ../../main.c:1042

`

EmergReanimator, 2025-03-23:
The same issue is observed with CPY 9.2.6;

I had to apply the patch below first:

```patch
diff --git a/samd/dma.c b/samd/dma.c
index 4925942..50a2cdb 100644
--- a/samd/dma.c
+++ b/samd/dma.c
@@ -64,9 +64,12 @@ uint8_t dma_allocate_channel(bool audio_channel) {
 }
 
 void dma_free_channel(uint8_t channel) {
-    assert(dma_allocated[channel]);
-    dma_disable_channel(channel);
-    dma_allocated[channel] = false;
+    if (dma_allocated[channel]) {
+        dma_disable_channel(channel);
+        dma_allocated[channel] = false;
+    }
+
+    return;
 }
```

```
#0  memcpy (dst=dst@entry=0x2003fb4c, src=src@entry=0x4000000, n=<optimized out>, n@entry=1) at ../../shared/libc/string0.c:78
#1  0x0004db04 in spi_flash_read_command (command=<optimized out>, command@entry=5 '\005', response=response@entry=0x2003fb4c "\002", length=length@entry=1) at supervisor/qspi_flash.c:57
#2  0x00058396 in wait_for_flash_ready () at ../../supervisor/shared/external_flash/external_flash.c:57
#3  0x000585d0 in write_flash (address=0, data=0x200033bc "\353\376\220MSDOS5.0", data_length=data_length@entry=256) at ../../supervisor/shared/external_flash/external_flash.c:104
#4  0x0004da86 in flush_ram_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:424
#5  spi_flash_flush_keep_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:446
#6  0x0005062e in supervisor_external_flash_flush () at ../../supervisor/shared/external_flash/external_flash.c:455
#7  supervisor_flash_flush () at ../../supervisor/shared/flash.c:180
#8  0x00029dce in external_flash_write_block (data=<optimized out>, block=8) at ../../supervisor/shared/external_flash/external_flash.c:521
#9  supervisor_flash_write_blocks (src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/external_flash/external_flash.c:556
#10 flash_write_blocks (self_in=<optimized out>, src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/flash.c:172
#11 flash_write_blocks (self_in=<optimized out>, src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/flash.c:154
#12 0x00024ac6 in mp_vfs_blockdev_write (self=<optimized out>, block_num=9, num_blocks=1, buf=<optimized out>) at ../../extmod/vfs_blockdev.c:118
#13 disk_write (pdrv=<optimized out>, buff=<optimized out>, sector=9, count=1) at ../../extmod/vfs_fat_diskio.c:86
#14 0x0002861e in f_mkfs.constprop.0 (fs=<optimized out>, opt=opt@entry=7 '\a', work=work@entry=0x2003fd50, len=512, au=<optimized out>) at ../../lib/oofatfs/ff.c:5852
#15 0x000291a8 in filesystem_init (create_allowed=true, force_create=false) at ../../supervisor/shared/filesystem.c:111
#16 0x0004df04 in main () at ../../main.c:1035
```

eightycc, 2025-03-23:
@EmergReanimator CircuitPython is built and tested with the GCC 13.3.Rel1 toolchain found here: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads. This toolchain includes patches specific to ARM cores. Is this the source of the toolchain you are using?

dhalbert, 2025-03-23:
So maybe `dma_free_channel()` is being called when the channel is already freed? We don't know whether the assertion is failing or not, but I think the assertion may be suppressed when not doing a debug build. We'll need to study the code to see whether `dma_free_channel()` should be idempotent or not.

Another possibility, but maybe less likely in this case.
This kind of problem in the past has been due to something not being `volatile` or to not putting something in a critical sectin.

In `peripherals/samd/sam_d5x_e5x/dma.c`, `dma_disable_channel()`, called above, is not in a disable-interrupts critical section. In `peripherals/samd/samd21/dma.c`, it is.

dhalbert, 2025-03-23:
@EmergReanimator How did you arrrive at that patch?

EmergReanimator, 2025-03-23:
> So maybe `dma_free_channel()` is being called when the channel is already freed? 

It does assert during the initialisation when built with DEBUG=1. 

> How did you arrrive at that patch?
This was the reason for my patch.
But this is only the case with latest 9.2.x (namely 9.2.6);

```
#0  __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at ../../main.c:1205
#1  0x000263e4 in dma_free_channel (channel=<optimized out>) at peripherals/samd/dma.c:66
#2  dma_free_channel (channel=<optimized out>) at peripherals/samd/dma.c:66
#3  0x0004a738 in audio_dma_reset () at audio_dma.c:337
#4  reset_port () at supervisor/port.c:368
#5  0x0004de6c in port_init () at supervisor/port.c:342
#6  main () at ../../main.c:993
```

dhalbert, 2025-03-23:
> This was the reason for my patch. But this is only the case with latest 9.2.x (namely 9.2.6);

With gcc 14 only, or also with gcc 13?

EmergReanimator, 2025-03-23:
> > This was the reason for my patch. But this is only the case with latest 9.2.x (namely 9.2.6);
> 
> With gcc 14 only, or also with gcc 13?

In fact it is another compiler independent issue:
`audio_dma_reset` calls `dma_free_channel` during the initialisation and `dma_free_channel` asserts in debug build only.

eightycc, 2025-03-23:
The GDB traceback looks like it might be a GCC code gen problem, which is why I asked about the toolchain. The patches from ARM sometimes take quite a while to get upstream into a GCC release.

EmergReanimator, 2025-03-23:
> [@EmergReanimator](https://github.com/EmergReanimator) CircuitPython is built and tested with the GCC 13.3.Rel1 toolchain found here: [...]

I have used arm-none-eabi-gcc (Fedora 14.1.0-2.fc41) 14.1.0 by default (GCC14).

I have just tried the ARM GCC 13.3 and CPY 9.2.6 without success. The CPY hangs in `filesystem_init`:
```
#0  memcpy (dst=dst@entry=0x2003fb3c, src=src@entry=0x4000000, n=n@entry=1) at ../../shared/libc/string0.c:45
#1  0x0004c6ec in spi_flash_read_command (command=<optimized out>, command@entry=5 '\005', response=response@entry=0x2003fb3c "\002", length=length@entry=1) at supervisor/qspi_flash.c:57
#2  0x00056e1e in wait_for_flash_ready () at ../../supervisor/shared/external_flash/external_flash.c:57
#3  0x0005705e in write_flash (address=0, data=0x200033bc "\353\376\220MSDOS5.0", data_length=data_length@entry=256) at ../../supervisor/shared/external_flash/external_flash.c:104
#4  0x0004c66e in flush_ram_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:424
#5  spi_flash_flush_keep_cache (keep_cache=true) at ../../supervisor/shared/external_flash/external_flash.c:446
#6  0x0004f1da in supervisor_external_flash_flush () at ../../supervisor/shared/external_flash/external_flash.c:455
#7  supervisor_flash_flush () at ../../supervisor/shared/flash.c:180
#8  0x00029092 in external_flash_write_block (block=8, data=0x2003fd40 "") at ../../supervisor/shared/external_flash/external_flash.c:521
#9  supervisor_flash_write_blocks (num_blocks=<optimized out>, block_num=<optimized out>, src=<optimized out>) at ../../supervisor/shared/external_flash/external_flash.c:556
#10 flash_write_blocks (self_in=<optimized out>, num_blocks=<optimized out>, block_num=<optimized out>, src=<optimized out>) at ../../supervisor/shared/flash.c:172
#11 flash_write_blocks (self_in=<optimized out>, src=<optimized out>, block_num=<optimized out>, num_blocks=<optimized out>) at ../../supervisor/shared/flash.c:154
#12 0x00023e66 in mp_vfs_blockdev_write (buf=<optimized out>, num_blocks=1, block_num=9, self=<optimized out>) at ../../extmod/vfs_blockdev.c:118
#13 disk_write (pdrv=<optimized out>, buff=<optimized out>, sector=9, count=1) at ../../extmod/vfs_fat_diskio.c:86
#14 0x000278dc in f_mkfs.constprop.0 (fs=<optimized out>, opt=opt@entry=7 '\a', work=work@entry=0x2003fd40, len=512, au=<optimized out>) at ../../lib/oofatfs/ff.c:5852
#15 0x00028478 in filesystem_init (create_allowed=true, force_create=false) at ../../supervisor/shared/filesystem.c:111
#16 0x0004cb00 in main () at ../../supervisor/shared/safe_mode.c:31
```

eightycc, 2025-03-23:
@EmergReanimator Thanks for verifying that!

EmergReanimator, 2025-03-23:
Here is the short summary of my results.
Unfortunately I could not get CPY 9.0.0 running any more.

| | ARM GCC10 | ARM GCC13 | GCC14 |
| -- | -- | -- | -- |
| 8.0.0 | :white_check_mark: | :no_entry: | :no_entry: |
| 9.0.0 | :exclamation: | :x:|:x: |
| 9.2.6 | :no_entry: | :x: | :x: |

:exclamation: - unstable, succeeded just once
:x: - hangs in filesystem initialisation
:no_entry: - build failure

---
* arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
* arm-none-eabi-gcc (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614
* arm-none-eabi-gcc (Fedora 14.1.0-2.fc41) 14.1.0

EmergReanimator, 2025-03-23:
FYI

CPY 9.0.0 runs fine with FS stubs built with GCC 14:
`Adafruit CircuitPython 9.0.0-dirty on 2025-03-23; SAM E54 Xplained Pro with same54p20`

``` patch
diff --git a/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk b/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
index d5038030e4..7263398e37 100644
--- a/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
+++ b/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
@@ -6,9 +6,12 @@ USB_MANUFACTURER = "Microchip"
 CHIP_VARIANT = SAME54P20A
 CHIP_FAMILY = same54
 
-QSPI_FLASH_FILESYSTEM = 1
-EXTERNAL_FLASH_DEVICES = "N25Q256A,SST26VF064B"
+# QSPI_FLASH_FILESYSTEM = 1
+# EXTERNAL_FLASH_DEVICES = "N25Q256A,SST26VF064B"
 LONGINT_IMPL = MPZ
 
 CIRCUITPY_SDIOIO = 1
 CIRCUITPY_CANIO = 1
+CIRCUITPY_FULL_BUILD = 0
+DISABLE_FILESYSTEM = 1
+CIRCUITPY_USB_MSC = 0

```

eightycc, 2025-03-23:
@EmergReanimator Very useful information throughout. I'm focusing on the flash filesystem code but will be delayed a few days waiting for some test parts to arrive.

eightycc, 2025-03-24:
@EmergReanimator Diff'ing the 8.1.0 and 8.2.0 tags I see the SST26VF064B was added to the list of possible flash parts. What is the flash part # on your board?

eightycc, 2025-03-24:
Possibly same as #8047. The file `data/nvm.toml/flash/microchip/SST26VF064B.toml` may be in error.

eightycc, 2025-03-24:
The SST26VF064B part requires that the IOC bit in its configuration register be set to 1 before quad read/write operations will succeed. There are two flavors of the SST26VF064, one with a B suffix the other with a BA suffix. The B suffix parts (the one shown in the photo in #8047) POR with this bit set to 0. I'm not finding any support for this bit in `check_quad_enable()`.

jepler, 2025-03-25:
check_quad_enable can write two bytes to the status register, which seems to be how the configuration register is written (see 5.30 in [the datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/MPD/ProductDocuments/DataSheets/SST26VF064B-SST26VF064BA-2-5V-3-0V-64-Mbit-Serial-Quad-IO--SQI--Flash-Memory-20005119K.pdf)). I think this'll make it write 00 02 which should set IOC:
```
quad_enable_bit_mask = 0x02
quad_enable_status_byte = 2
```
(to be updated in `data/nvm.toml/flash/microchip/SST26VF064B.toml`)

total guess, didn't test anything.

eightycc, 2025-03-25:
@jepler That might work. Thanks for taking the time to have a look. I'll desk check and PR if it looks OK.

eightycc, 2025-03-25:
@jepler 's suggestion desk checks. `quad_enable_status_byte = 2` causes `single_status_byte = false` after the `.toml` is ninja'ed, which is exactly what we want. 

The only crunchy bit is in `supervisor_flash_init()`:https://github.com/adafruit/circuitpython/blob/a0b482c2806ac19962ed26f2d4bbe5ebdef3406a/supervisor/shared/external_flash/external_flash.c#L238-L248
Here we'll read the configuration register at lines 245-246 and test the `WPEN` (Write Protection Pin Enable). This should always be 0, so although we're not actually testing for a suspended write/erase, we should fall through.

eightycc, 2025-03-25:
@EmergReanimator Please make the change noted above by @jepler to `data/nvm.toml/flash/microchip/SST26VF064B.toml` and re-test. I'm assuming you've got a board stuffed with the `SST26VF064B` part.

EmergReanimator, 2025-03-25:
> [@EmergReanimator](https://github.com/EmergReanimator) Diff'ing the 8.1.0 and 8.2.0 tags I see the SST26VF064B was added to the list of possible flash parts. What is the flash part # on your board?

I have got a SAME54 Xplained board with 26F064B flash.

> [@EmergReanimator](https://github.com/EmergReanimator) Please make the change noted above by [@jepler](https://github.com/jepler) to `data/nvm.toml/flash/microchip/SST26VF064B.toml` and re-test. I'm assuming you've got a board stuffed with the `SST26VF064B` part.

The commit https://github.com/adafruit/nvm.toml/pull/20/commits/9dc99a20244d7d06e207b1f329783da1f4935f09 did not help.

Oddly enough, but the 8.2.9 boots fine when external flash is set to `N25Q256A`;

`Adafruit CircuitPython 8.2.9-dirty on 2025-03-25; SAM E54 Xplained Pro with same54p20`

``` patch
diff --git a/data/nvm.toml b/data/nvm.toml
--- a/data/nvm.toml
+++ b/data/nvm.toml
@@ -1 +1 @@
-Subproject commit 6b678f15e378edce820f2ffdef3286b3e55449e7
+Subproject commit 6b678f15e378edce820f2ffdef3286b3e55449e7-dirty
diff --git a/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk b/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
index d5038030e4..53aac7df2a 100644
--- a/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
+++ b/ports/atmel-samd/boards/same54_xplained/mpconfigboard.mk
@@ -7,7 +7,7 @@ CHIP_VARIANT = SAME54P20A
 CHIP_FAMILY = same54
 
 QSPI_FLASH_FILESYSTEM = 1
-EXTERNAL_FLASH_DEVICES = "N25Q256A,SST26VF064B"
+EXTERNAL_FLASH_DEVICES = "N25Q256A"
 LONGINT_IMPL = MPZ
 
 CIRCUITPY_SDIOIO = 1
```

eightycc, 2025-03-25:
@EmergReanimator Thank you for your patience and persistence. I've updated the `.toml` PR, but as I noted there the update does not change the `jinja`-generated `genhdr/devices.h`.

I suspect there's a deeper problem in `atmel-samd`-specific flash support code. I will pick this up again later in the week when my hardware order arrives.

EmergReanimator, 2025-03-28:
> [@EmergReanimator](https://github.com/EmergReanimator) Thank you for your patience and persistence. I've updated the `.toml` PR, but as I noted there the update does not change the `jinja`-generated [...]

> // Wait until both the write enable and write in progress bits have cleared.

Why the flash is ready when the WEL is cleared?
I left `WEL` bit out and kept the `BUSY` polling and now it works correctly.

``` patch
diff --git a/supervisor/shared/external_flash/external_flash.c b/supervisor/shared/external_flash/external_flash.c
index f72518ac50..f505408786 100644
--- a/supervisor/shared/external_flash/external_flash.c
+++ b/supervisor/shared/external_flash/external_flash.c
@@ -70,7 +70,7 @@ static bool wait_for_flash_ready(void) {
     uint8_t read_status_response[1] = {0x00};
     do {
         ok = spi_flash_read_command(CMD_READ_STATUS, read_status_response, 1);
-    } while (ok && (read_status_response[0] & 0x3) != 0);
+    } while (ok && (read_status_response[0] & 0x1) != 0x0);
     return ok;
 }

```

eightycc, 2025-03-28:
That code bothered me, too. Are you running with this as the only change? Does the CIRCUITPY device mount with a size < 8MB?

eightycc, 2025-03-28:
>Why the flash is ready when the WEL is cleared?

The code is intended to be generic for all of the flash devices CP supports. I suspect it's there for one or more of the other 600+ boards we support. I read it as handling a reset when the flash chip was busy with a lengthy programming operation.

EmergReanimator, 2025-03-28:
> Does the CIRCUITPY device mount with a size < 8MB?

8.4 MiB to be precise.

```
[11205.487427] usb 2-7.1.2: new full-speed USB device number 28 using xhci_hcd
[11205.573863] usb 2-7.1.2: New USB device found, idVendor=239a, idProduct=80b6, bcdDevice= 1.00
[11205.573873] usb 2-7.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11205.573878] usb 2-7.1.2: Product: SAM E54 Xplained Pro
[11205.573881] usb 2-7.1.2: Manufacturer: Microchip
[11205.573885] usb 2-7.1.2: SerialNumber: 0AAA3C345943385320202046431718FF
[11205.593543] cdc_acm 2-7.1.2:1.0: ttyACM0: USB ACM device
[11205.594212] usb-storage 2-7.1.2:1.2: USB Mass Storage device detected
[11205.594720] scsi host10: usb-storage 2-7.1.2:1.2
[11205.597114] input: Microchip SAM E54 Xplained Pro Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-7/2-7.1/2-7.1.2/2-7.1.2:1.3/0003:239A:80B6.0006/input/input27
[11205.687678] input: Microchip SAM E54 Xplained Pro Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-7/2-7.1/2-7.1.2/2-7.1.2:1.3/0003:239A:80B6.0006/input/input28
[11205.688014] hid-generic 0003:239A:80B6.0006: input,hidraw0: USB HID v1.11 Keyboard [Microchip SAM E54 Xplained Pro] on usb-0000:00:14.0-7.1.2/input3
[11205.688871] usb 2-7.1.2: Quirk or no altset; falling back to MIDI 1.0
[11206.650193] scsi 10:0:0:0: Direct-Access     Microchi SAM E54 Xplained 1.0  PQ: 0 ANSI: 2
[11206.650851] sd 10:0:0:0: Attached scsi generic sg4 type 0
[11206.652219] sd 10:0:0:0: [sdd] 16377 512-byte logical blocks: (8.39 MB/8.00 MiB)
[11206.653120] sd 10:0:0:0: [sdd] Write Protect is off
[11206.653125] sd 10:0:0:0: [sdd] Mode Sense: 03 00 00 00
[11206.654540] sd 10:0:0:0: [sdd] No Caching mode page found
[11206.654545] sd 10:0:0:0: [sdd] Assuming drive cache: write through
[11206.681647]  sdd: sdd1
[11206.681854] sd 10:0:0:0: [sdd] Attached SCSI removable disk
```

EmergReanimator, 2025-03-28:
> That code bothered me, too. Are you running with this as the only change? 

That one too.
https://github.com/adafruit/nvm.toml/pull/20/commits/26148d1b00973f49a1d6a843f42f1aea5a43d846

eightycc, 2025-03-28:
Perfect. I'll PR a more general solution later that includes the `nvm.toml` update. If I'm not mistaken, you're unblocked for now.

EmergReanimator, 2025-03-28:
> Perfect. I'll PR a more general solution later that includes the `nvm.toml` update. If I'm not mistaken, you're unblocked for now.

Yes, you can say so. @eightycc  Thanks for you eager support!

Just for your information:
I am maintaining NXP port of the CPY. It was a while since NXP-fork was synchronised with upstream - CPY V8.0;
I was going to updated NXP port fork to 9.x; But first I tried the `same54_xplained` board before adding NXP port to 9.x.

EmergReanimator, 2025-03-30:
> I left WEL bit out and kept the BUSY polling and _now it works correctly_.

I have to revise my statement: the reading seems to work correctly.
Not able to store anything or modify the FAT filesystem in any way.

eightycc, 2025-03-31:
Are the failing writes from code running within CP, or to the mounted MSC filesystem, or both? What is the behavior you're seeing when you try to write?

EmergReanimator, 2025-03-31:
> Are the failing writes from code running within CP, or to the mounted MSC filesystem, or both? What is the behavior you're seeing when you try to write?

Sorry, forgot to note  it in my previous comment.
There is still an issue in external flash code. It does not work well with `SST26VF064B`.
But do work correctly with `N25Q256A`.

I have my older _SAME54 Xplained_ board with `N25Q256A` flash back. The filesystem works properly with the same binary on that board.

eightycc, 2025-04-11:
I've identified two root causes that are causing this bug:

1. Failure to set the IOC bit in the flash configuration register was causing the part's WP# and HOLD# pins to be incorrectly configured for quad SPI operation. Additionally, a bug in CP flash initialization was causing the IOC bit to be reset after it was set.
2. Sector protection scheme unique to this flash part was resulting in quiet failure of sector erase commands. A special ULBPR command is necessary during flash initialization to disable sector protection.

Tested with experimental changes. CIRCUITPY device now initializes and mounts r/w correctly on a red same54_xplained board with the SST26VF064B part.

A PR will follow shortly once I've worked out a way to introduce these changes in a way that is compatible with the other flash parts that CP supports.

eightycc, 2025-04-14:
@EmergReanimator Please re-test with the build artifact from #10249. Thank you!

closed 2025-04-14 by tannewt

EmergReanimator, 2025-04-15:
> [@EmergReanimator](https://github.com/EmergReanimator) Please re-test with the build artifact from [#10249](https://github.com/adafruit/circuitpython/pull/10249). Thank you!

Yup. Now it works correctly: 
```
Adafruit CircuitPython 10.0.0-alpha.2-14-g52336ef192-dirty on 2025-04-15; SAM E54 Xplained Pro with same54p20
Board ID:same54_xplained
UID:96A8371F32543753202020392D1C04FF
```

```
Adafruit CircuitPython 10.0.0-alpha.2-14-g52336ef192-dirty on 2025-04-15; SAM E54 Xplained Pro with same54p20
Board ID:same54_xplained
UID:0AAA3C345943385320202046431718FF
```