adafruit/circuitpython #10175: Distorted mp3 playback on MAX98357A

bbtinkerer, 2025-03-24:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.6 on 2025-03-23; Raspberry Pi Pico 2 with rp2350a
Board ID:raspberry_pi_pico2

same issue when using Pi Pico rp2040
```

### Code/REPL

```python
# sample code and mp3 from https://learn.adafruit.com/mp3-playback-rp2040/pico-i2s-mp3
#
# SPDX-FileCopyrightText: 2021 Kattni Rembor for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""
CircuitPython I2S MP3 playback example.
Plays a single MP3 once.
"""
import board
import audiomp3
import audiobusio

audio = audiobusio.I2SOut(board.GP0, board.GP1, board.GP2)

mp3 = audiomp3.MP3Decoder(open("slow.mp3", "rb"))

audio.play(mp3)
while audio.playing:
    pass

print("Done playing!")
```

### Behavior

Audio playback of mp3s are distorted when playing back using a MAX98357A. 

### Description

Distorted playback also with CircuitPython 9.2.5 on both Pi Pico with rp2040 and Pi Pico 2 rp2350

### Additional information

Tried using audiomixer to lower volume since audio sounded like a high gain was applied to it, but that only lowered the volume, the audio was still distorted.

MP3s play fine when using CircuitPython 9.2.4 and earlier.

bablokb, 2025-03-24:
I can confirm this behavior using a PCM5102 and a Pico2-W. So this is not related to the specific I2S-amp chip MAX98357A.

bablokb, 2025-03-24:
It is not related to the Pico, I have the same issue with an ESP32-S3

jepler, 2025-03-26:
I reproduced the same/similar issue on pygamer. Through git bisect, I narrowed it down to e15eba0d33f4b09490d5d4ce2431b5bd8e225e16 (#10036).

I'll investigate further & hopefully make a fix.

bablokb, 2025-03-27:
PR #10185 fixes it for me. Tested with a Pico2-W and PCM5102.

Thanks!

jepler, 2025-03-27:
thanks for testing @bablokb !

dhalbert, 2025-03-27:
Fixed by #10185.

closed 2025-03-27 by dhalbert