adafruit/circuitpython #10200: synthio filters glitch at certain frequencies with Q > 1.41

todbot, 2025-03-29:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.6 on 2025-03-23; Raspberry Pi Pico 2 with rp2350a
Adafruit CircuitPython 9.2.6 on 2025-03-23; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
"""
filter Q over 1.41 causes glitches in the filter at certain frequency
"""

import time, board, synthio, audiobusio, audiomixer

SAMPLE_RATE = 44100
BUFFER_SIZE = 2048
i2s_bck_pin = board.GP20
i2s_lck_pin = board.GP21
i2s_dat_pin = board.GP22

audio = audiobusio.I2SOut(bit_clock=i2s_bck_pin, word_select=i2s_lck_pin, data=i2s_dat_pin)
mixer = audiomixer.Mixer(sample_rate=SAMPLE_RATE, channel_count=2, buffer_size=BUFFER_SIZE)
synth = synthio.Synthesizer(sample_rate=SAMPLE_RATE, channel_count=2)
audio.play(mixer)
mixer.voice[0].play(synth)

midi_notes = [ 49,  # glitch near 170 & 150 Hz
               37,  # glitch near 100, 70, & 60 Hz
               36,  # gitch near 70 & 60 Hz
               42,  # glitch near 100 Hz
               47,  # glitch near 120 Hz
              ]
while True:
    for n in midi_notes:
        midi_note_f = synthio.midi_to_hz(n)
        note = synthio.Note(midi_note_f)
        note.filter = synthio.BlockBiquad(synthio.FilterMode.LOW_PASS,
                                          frequency=200, Q=1.5)
        synth.press(note)
        for f in range(21):  # modulate the filter down by hand to find glitches
            filter_freq = 200 - (10*f)
            print("Q: %.2f filter freq: %3d note freq: %3.2f" %
                  (note.filter.Q, filter_freq, midi_note_f))
            note.filter.frequency = filter_freq
            time.sleep(0.3)        
        synth.release(note)
```

### Behavior

When `synthio.BlockBiquad` is set to certain frequencies with Q>1.41, strange oscillations / "glitches" occur that should not. In general it seems to be around the pitch of the playing note and lower. 

### Description

Here's a video showing the above code running on a Pico2 RP2350 driving an I2S DAC.
The behavior is the same on a Pico RP2040 with I2S DAC. The behavior is also the same with `PWMOut`.
This behavior has existed since `synthio` has had filters, but with `BlockBiquad` it's more noticeable because we can easily modulate the filter frequency.

https://github.com/user-attachments/assets/26b8127b-7dfd-4ef1-9536-73c93c4421b3

gamblor21, 2025-04-21:
Took a quick look, the waveform (in Audacity) looks like a overflow error. Quickly traced it back and in:
 https://github.com/adafruit/circuitpython/blob/ceda2f0545d0b16261563aeef92f4bef576bf01d/shared-module/synthio/__init__.c#L272 

there is a likely overflow error (int32_t * int16_t into a int32_t) that Jeff noticed when we were discussing fixed point. I think this is the same error that was causing some random distortion I was getting with louder volumes and multiple voices.

gamblor21, 2025-04-21:
> I also hear distortion at times with many voices that I bet it's due to this. So if we find a good cheap fixed-point saturation algorithm, maybe we put it in `sum_with_loudness()`.

I use one already in Reverb so when I have a moment I'm going to look through `synthio` and try to catch places where we are letting things get too large. Probably should be its own issue but your report is easy to reproduce. I never could find a magic way.

todbot, 2025-04-21:
I've not looked, but it seems that whatever `audiofilters.Filter` is doing is different.  That is, by moving using an `audiofilter` instead of a per-Note filter like below, the glitches are gone (but there are odd bumps in volume at the glitch frequencies)

```py
""" .... set up audio system including a mixer and synth """

filter1 = audiofilters.Filter(buffer_size=1024,
                              channel_count=CHANNEL_COUNT,
                              sample_rate=SAMPLE_RATE,
                              mix=1.0)
filter1.filter = synthio.Biquad(synthio.FilterMode.LOW_PASS,
                         frequency=200, Q=1.5)
mixer.voice[0].play(filter1)
filter1.play(synth)

midi_notes = [ 49,  # glitch near 170 & 150 Hz
               37,  # loud near 100, 70, & 60 Hz
               36,  # loud near 70 & 60 Hz
               42,  # loud near 100 Hz
               47,  # loud near 120 Hz
              ]

while True:
    for n in midi_notes:
        midi_note_f = synthio.midi_to_hz(n)
        note = synthio.Note(midi_note_f)
        synth.press(note)
        for f in range(21):  # modulate the filter down by hand to find glitches
            filter_freq = 200 - (10*f)
            print("Q: %.2f filter freq: %3d note freq: %3.2f" %
                  (filter1.filter.Q, filter_freq, midi_note_f))
            filter1.filter.frequency = filter_freq
            time.sleep(0.3)        
        synth.release(note)
```

relic-se, 2025-04-22:
Thank you for researching into this, @todbot! I think all of us synthio folks have been aware of this problem for a while. I'd love to push the filter harder, but it doesn't take long to run into this error. I'm sure that `audiofilters.Filter` is implementing summing slightly better, but I'd be very interested to see what your research within `audiofreeverb.Freeverb` can do for this issue, @gamblor21.

gamblor21, 2025-04-23:
#10288 fixed this for me @todbot if you have a chance I'd love you to give it a try.

todbot, 2025-04-25:
Oh my goodness, yes! Thank you. The #10288 patch makes the filter distortion non-existent! Much easier to do chords other other multi-note stuff now too.

gamblor21, 2025-04-28:
Fixed by #10288