adafruit/circuitpython #10201: Adafruit Metro RP2350 analogbufio

mikeysklar, 2025-03-30:
### CircuitPython version and board name

```python
Circuit 9.2.6
```

### Code/REPL

```python
import board
import analogbufio
import array

length = 1000
mybuffer = array.array("H", [0x0000] * length)
rate = 500000
adcbuf = analogbufio.BufferedIn(board.A0, sample_rate=rate)
adcbuf.readinto(mybuffer)
adcbuf.deinit()
for i in range(length):
    print(i, mybuffer[i])
```

### Behavior

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
ValueError: Invalid pins
```

### Description

Metro RP2350 fails to run analogbufio example code [from the docs](https://docs.circuitpython.org/en/latest/shared-bindings/analogbufio/index.html). 

Code works as expected on Feather RP2040, Feather RP2350 or switching to analogio.

### Additional information

[Forum issue](https://forums.adafruit.com/viewtopic.php?t=217577) found by mattblair

dhalbert, 2025-03-30:
Thanks - I saw that forum post this morning and am working on a fix. It's due to not treating the RP2350B pins properly.

dhalbert, 2025-04-01:
Fixed by #10202.

PR #10202 (2025-03-31): Fix analogbufio.BufferedIn pin validation on RP2350B
- Fixes #10201.

- Fix `analogbufio.BufferedIn` pin validation on RP2350B.
- Use`pico-sdk`-supplied `ADC_BASE_PIN` instead of figuring out the base pin ourselves.
- Disallow using `board.A3` (GPIO29) on Pico W and Pico 2 W for `analogbufio.BufferedIn`, and document that. GPIO29  pin has a voltage divider attached to monitor VSYS, and is also shared with SPI SCK connected to the radio module. `AnalogIn` special-cases `A3` to make it available, but that was complicated, and I think the use case for GPIO29 with `analogbufio` is limited.
closed 2025-04-01 by dhalbert