adafruit/circuitpython #10203: audiofilters.Filter fails when assign 'filter' property

todbot, 2025-03-30:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.6 on 2025-03-23; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
import time, board, audiobusio, synthio, audiofilters

audio = audiobusio.I2SOut(bit_clock=board.GP20, word_select=board.GP21, data=board.GP22)
synth = synthio.Synthesizer(channel_count=1, sample_rate=44100)
effect = audiofilters.Filter(buffer_size=1024, channel_count=1, sample_rate=44100, mix=1.0)
effect.filter = synth.low_pass_filter(frequency=2000, Q=1.25)
effect.play(synth)
audio.play(effect)

note = synthio.Note(261)
while True:
    synth.press(note)
    time.sleep(0.25)
    synth.release(note)
    time.sleep(5)
```

### Behavior

```
Traceback (most recent call last):
  File "code.py", line 10, in <module>
TypeError: object in filter must be of type AnyBiquad, not float
```

### Description

The above code is from the demo code in documentation for `audiofillters.Filter` https://docs.circuitpython.org/en/latest/shared-bindings/audiofilters/index.html#audiofilters.Filter

todbot, 2025-04-04:
Fixed by (I think) commit 05a50b8fd8f9146b0fe9dea28e55182c9acb4f5b.  Thanks @relic-se!

closed 2025-04-04 by todbot

relic-se, 2025-04-04:
@todbot Tested this on my end with the new PR, and it works without error. The only other change is that in 10.x, you'll need to use `synthio.Biquad` instead of `synthio.BlockBiquad`.