adafruit/circuitpython #10206: Support raw f-strings

AbortRetryFail, 2025-03-31:
Please add support for raw f-strings.

``` python
>>> import glob
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/glob.py", line 319
SyntaxError: raw f-strings are not supported
```

It looks like MicroPython did this last year, and it didn't even make the code bigger. https://github.com/micropython/micropython/pull/15218

I'll try to apply this patch myself and see if it works, but I might be way out of my depth here.

AbortRetryFail, 2025-03-31:
I had to tweak their patch a bit, but it builds and seems to work OK on my LILYGO T-Deck.

dhalbert, 2025-03-31:
We periodically merge from upstream and will be including that change the next time, so if you submit a PR, try to make it as similar as possible, and let us know what you had to tweak

AbortRetryFail, 2025-03-31:
Well, I tried to attach the patch here but Github just says "failed to upload"

Doing it the hard way with the whole fork and PR song and dance.

AbortRetryFail, 2025-04-02:
I hit another, taller wall with what I was trying to do, so I stopped working on this. I guess close it out the next time you merge from MicroPython.

dhalbert, 2025-04-02:
Sorry to hear! But it's on the list -- thanks.