adafruit/circuitpython #10218: Allow Selection of _bleio by Board

eightycc, 2025-04-02:
From PR #9878 @dhalbert suggests:

Suppose `_bleio` for HCI was actually called `_bleio_hci`. The libraries would do something like:
```py
try:
    import _bleio
except ImportError
    import _bleio_hci as _bleio
```
Conceivably, one could make a build that would support both, though it's not of much use (though there are compile issues about that, so you don't have to duplicate the shared-bindings code).

dhalbert, 2025-04-02:
I am not sure this is a good idea at all, and I'd say don't bother with a PR for this right now.

tannewt, 2025-04-04:
I'm also not sure this is the best way. I think in the future _bleio will build on one BLE host stack (Zephyr) that we can configure with an HCI transport. Native BLE will then make it available as `board.BLE` or something.

closed 2025-04-04 by dhalbert