adafruit/circuitpython #10224: alarm.exit_and_deep_sleep_until_alarms(pin_alarm) fails to deep sleep on Xiao ESP32 C6

timshac, 2025-04-05:
### CircuitPython version and board name

```python
Adafruit CircuitPython 9.2.6 on 2025-03-23; Seeed Xiao ESP32-C6 4MB Flash 512KB SRAM with ESP32-C6FH4
```

### Code/REPL

```python
import board
import alarm
import microcontroller
# LED pin for observing if deep sleep was reached (LED will go out)
LED_PIN = microcontroller.pin.GPIO15
led = digitalio.DigitalInOut(LED_PIN)
led.direction = digitalio.Direction.OUTPUT
led.value = False  # turn on LED
pin_alarm = alarm.pin.PinAlarm(pin=board.D9, value=True, pull=False)
alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
# board does not deep sleep here
# this code (below) will correctly sleep, forever:
# alarm.exit_and_deep_sleep_until_alarms()
```

### Behavior

With the pin_alarm, the board will not sleep
```python
alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
```
The device will not deep sleep.

With no alarms, the board will sleep forever
```python
alarm.exit_and_deep_sleep_until_alarms()
```
This code will correctly deep sleep.

Deep sleeping correctly works with a time alarm:
```python
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 10)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```
This code will correctly deep sleep, and wake up 10s later.

dhalbert, 2025-04-07:
If you jumper D9 (GPIO20) to ground, does it sleep? Is anything connected to D9?

timshac, 2025-04-07:
Good point.

If I strap D9 to ground or 3V3, the board still does not sleep.

I tried with the following alarms individually, all which failed to sleep:
```python
pin_alarm = alarm.pin.PinAlarm(pin=board.D9, value=True, pull=False)
pin_alarm = alarm.pin.PinAlarm(pin=board.D9, value=False, pull=True)
pin_alarm = alarm.pin.PinAlarm(pin=board.D9, value=False)
pin_alarm = alarm.pin.PinAlarm(pin=board.D9, value=True) 
```

eightycc, 2025-05-14:
For the ESP32-C6, only pins GPIO0 through GPIO7 may be used for alarms. In low-power states, these are the only pins whose pads are powered. CircuitPython should be throwing a value error on the attempt to use GPIO20 as an alarm. From section 12.4 of "ESP32-C6 Technical Resource Manual":

> Eight LP GPIO pins (GPIO0 ∼ GPIO7): These pins are always powered up and are not affected by any low-power modes, which makes them suitable for working as wake-up sources when the chip is in low-power modes. These pins can also work as regular GPIOs.

Attempting to use one of the LP supported pins (in this case A0 (GPIO1 on the Adafruit Feather ESP32C6) results in:
```
>>> pin_alarm = alarm.pin.PinAlarm(pin=board.A0, value=True, pull=False)
>>> alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Can only alarm on RTC IO from deep sleep.
```

eightycc, 2025-05-16:
The provided example is attempting deep sleep on GPIO20, a pin that is not supported for that purpose on the ESP32-C6. Attempting to use an unsupported pin on an Adafruit Feather ESP32C6 correctly results in a `Value Error` like this `code.py`:
```python
import board
import alarm
import microcontroller
import digitalio
import time

LED_PIN = microcontroller.pin.GPIO15
led = digitalio.DigitalInOut(LED_PIN)
led.direction = digitalio.Direction.OUTPUT
led.value = True

time.sleep(10)
pin_alarm = alarm.pin.PinAlarm(pin=board.IO14, value=False, pull=False)
print("entering deep sleep...")
alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
```
```
code.py output:
entering deep sleep...
Traceback (most recent call last):
  File "code.py", line 15, in <module>
ValueError: Can only alarm on RTC IO from deep sleep.

Code done running.
```
Changing the pin from `IO14` to `IO1`:
```python
pin_alarm = alarm.pin.PinAlarm(pin=board.IO1, value=False, pull=False)
```
while wiring a push button with a 10K pull-up to 3.3V and ground when pressed results in deep sleep until the button is pressed.

@timshac Please re-test using a supported pin.

Something that threw me off while testing: Entering deep sleep is not supported from REPL. This includes attempting to run a Python script by `import test_deep` or typing `alarm.exit_and_deep_sleep_until_alarms...`. This should be documented or corrected.

timshac, 2025-05-16:
```
import board
import time
import alarm
import time
import digitalio
import microcontroller

LED_PIN = microcontroller.pin.GPIO15
led = digitalio.DigitalInOut(LED_PIN)
m_d8 = microcontroller.pin.GPIO19
m_d9 = microcontroller.pin.GPIO20
led.direction = digitalio.Direction.OUTPUT

for i in range(5):
    led.value = False  # turn on LED
    time.sleep(1)
    led.value = True  # turn off LED
    time.sleep(1)

pin_alarmD2 = alarm.pin.PinAlarm(pin=board.D2, value=False, pull=True) # doesn't work with 9.2.6 C6
alarm.exit_and_deep_sleep_until_alarms(pin_alarmD2)
```
*Board sleeps when not connected to repl.*

I can confirm that the board correctly sleeps using an LP_* pin. Thank you!

closed 2025-05-16 by eightycc