adafruit/circuitpython #10225: RP2040-Zero: Write Protect is on - Ubuntu Linux

iz8mbw, 2025-04-06:
### CircuitPython version and board name

```python
CircuitPython 9.2.7 RP2040-Zero
```

### Code/REPL

```python
import time
import board
import busio
import adafruit_scd4x
import adafruit_sht4x
import adafruit_sht31d

i2c = busio.I2C(board.GP1, board.GP0)
si2c = busio.I2C(board.GP3, board.GP2)
```

### Behavior

[144854.408519] usb 9-2: new full-speed USB device number 4 using xhci_hcd
[144854.539664] usb 9-2: New USB device found, idVendor=2e8a, idProduct=101f, bcdDevice= 1.00
[144854.539669] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[144854.539671] usb 9-2: Product: RP2040-Zero
[144854.539672] usb 9-2: Manufacturer: Waveshare Electronics
[144854.539674] usb 9-2: SerialNumber: XXXXXXXXXXXXXXXX
[144854.543214] cdc_acm 9-2:1.0: ttyACM0: USB ACM device
[144854.545048] usb-storage 9-2:1.2: USB Mass Storage device detected
[144854.545451] scsi host7: usb-storage 9-2:1.2
[144854.548951] input: Waveshare Electronics RP2040-Zero Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0003/input/input9
[144854.636780] input: Waveshare Electronics RP2040-Zero Mouse as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0003/input/input10
[144854.636956] hid-generic 0003:2E8A:101F.0003: input,hidraw0: USB HID v1.11 Keyboard [Waveshare Electronics RP2040-Zero] on usb-0000:07:1b.0-2/input3
[144854.638579] usb 9-2: Quirk or no altset; falling back to MIDI 1.0
[144855.582288] scsi 7:0:0:0: Direct-Access     Waveshar RP2040-Zero      1.0  PQ: 0 ANSI: 2
[144855.583276] sd 7:0:0:0: Attached scsi generic sg2 type 0
[144855.588629] sd 7:0:0:0: [sdb] 2049 512-byte logical blocks: (1.05 MB/1.00 MiB)
[144855.594409] sd 7:0:0:0: [sdb] Write Protect is on
[144855.594414] sd 7:0:0:0: [sdb] Mode Sense: 03 00 80 00
[144855.595934] sd 7:0:0:0: [sdb] No Caching mode page found
[144855.595938] sd 7:0:0:0: [sdb] Assuming drive cache: write through
[144855.610203]  sdb: sdb1
[144855.611726] sd 7:0:0:0: [sdb] Attached SCSI removable disk

### Description

Hello.
Using a RP2040-Zero on Linux Ubuntu 24.04.2 LTS (GNU/Linux 6.8.1-1019-realtime x86_64) and often when I connect the RP2040-Zero to USB the disk partition is in read only (Write Protect is on).

See this log about some plug and unplug of the RP2040-Zero to my machine:

```
[144829.465753] usb 9-2: new full-speed USB device number 3 using xhci_hcd
[144829.596618] usb 9-2: New USB device found, idVendor=2e8a, idProduct=101f, bcdDevice= 1.00
[144829.596624] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[144829.596625] usb 9-2: Product: RP2040-Zero
[144829.596627] usb 9-2: Manufacturer: Waveshare Electronics
[144829.596628] usb 9-2: SerialNumber: XXXXXXXXXXXXXXXX
[144829.600903] cdc_acm 9-2:1.0: ttyACM0: USB ACM device
[144829.601943] usb-storage 9-2:1.2: USB Mass Storage device detected
[144829.602123] scsi host7: usb-storage 9-2:1.2
[144829.605795] input: Waveshare Electronics RP2040-Zero Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0002/input/input7
[144829.697951] input: Waveshare Electronics RP2040-Zero Mouse as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0002/input/input8
[144829.698150] hid-generic 0003:2E8A:101F.0002: input,hidraw0: USB HID v1.11 Keyboard [Waveshare Electronics RP2040-Zero] on usb-0000:07:1b.0-2/input3
[144829.699709] usb 9-2: Quirk or no altset; falling back to MIDI 1.0
[144830.623142] scsi 7:0:0:0: Direct-Access     Waveshar RP2040-Zero      1.0  PQ: 0 ANSI: 2
[144830.624196] sd 7:0:0:0: Attached scsi generic sg2 type 0
[144830.627293] sd 7:0:0:0: [sdb] 2049 512-byte logical blocks: (1.05 MB/1.00 MiB)
[144830.628694] sd 7:0:0:0: [sdb] Write Protect is off
[144830.628699] sd 7:0:0:0: [sdb] Mode Sense: 03 00 00 00
[144830.630338] sd 7:0:0:0: [sdb] No Caching mode page found
[144830.630341] sd 7:0:0:0: [sdb] Assuming drive cache: write through
[144830.649779]  sdb: sdb1
[144830.650789] sd 7:0:0:0: [sdb] Attached SCSI removable disk
[144841.841194] usb 9-2: USB disconnect, device number 3
[144854.408519] usb 9-2: new full-speed USB device number 4 using xhci_hcd
[144854.539664] usb 9-2: New USB device found, idVendor=2e8a, idProduct=101f, bcdDevice= 1.00
[144854.539669] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[144854.539671] usb 9-2: Product: RP2040-Zero
[144854.539672] usb 9-2: Manufacturer: Waveshare Electronics
[144854.539674] usb 9-2: SerialNumber: XXXXXXXXXXXXXXXX
[144854.543214] cdc_acm 9-2:1.0: ttyACM0: USB ACM device
[144854.545048] usb-storage 9-2:1.2: USB Mass Storage device detected
[144854.545451] scsi host7: usb-storage 9-2:1.2
[144854.548951] input: Waveshare Electronics RP2040-Zero Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0003/input/input9
[144854.636780] input: Waveshare Electronics RP2040-Zero Mouse as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0003/input/input10
[144854.636956] hid-generic 0003:2E8A:101F.0003: input,hidraw0: USB HID v1.11 Keyboard [Waveshare Electronics RP2040-Zero] on usb-0000:07:1b.0-2/input3
[144854.638579] usb 9-2: Quirk or no altset; falling back to MIDI 1.0
[144855.582288] scsi 7:0:0:0: Direct-Access     Waveshar RP2040-Zero      1.0  PQ: 0 ANSI: 2
[144855.583276] sd 7:0:0:0: Attached scsi generic sg2 type 0
[144855.588629] sd 7:0:0:0: [sdb] 2049 512-byte logical blocks: (1.05 MB/1.00 MiB)
[144855.594409] sd 7:0:0:0: [sdb] Write Protect is on
[144855.594414] sd 7:0:0:0: [sdb] Mode Sense: 03 00 80 00
[144855.595934] sd 7:0:0:0: [sdb] No Caching mode page found
[144855.595938] sd 7:0:0:0: [sdb] Assuming drive cache: write through
[144855.610203]  sdb: sdb1
[144855.611726] sd 7:0:0:0: [sdb] Attached SCSI removable disk
[144865.872787] usb 9-2: USB disconnect, device number 4
[144872.418499] usb 9-2: new full-speed USB device number 5 using xhci_hcd
[144872.549227] usb 9-2: New USB device found, idVendor=2e8a, idProduct=101f, bcdDevice= 1.00
[144872.549233] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[144872.549234] usb 9-2: Product: RP2040-Zero
[144872.549236] usb 9-2: Manufacturer: Waveshare Electronics
[144872.549237] usb 9-2: SerialNumber: XXXXXXXXXXXXXXXX
[144872.555569] cdc_acm 9-2:1.0: ttyACM0: USB ACM device
[144872.556622] usb-storage 9-2:1.2: USB Mass Storage device detected
[144872.557930] scsi host7: usb-storage 9-2:1.2
[144872.560645] input: Waveshare Electronics RP2040-Zero Keyboard as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0004/input/input11
[144872.653613] input: Waveshare Electronics RP2040-Zero Mouse as /devices/pci0000:00/0000:00:1e.0/0000:05:02.0/0000:07:1b.0/usb9/9-2/9-2:1.3/0003:2E8A:101F.0004/input/input12
[144872.654025] hid-generic 0003:2E8A:101F.0004: input,hidraw0: USB HID v1.11 Keyboard [Waveshare Electronics RP2040-Zero] on usb-0000:07:1b.0-2/input3
[144872.655175] usb 9-2: Quirk or no altset; falling back to MIDI 1.0
[144873.566221] scsi 7:0:0:0: Direct-Access     Waveshar RP2040-Zero      1.0  PQ: 0 ANSI: 2
[144873.567007] sd 7:0:0:0: Attached scsi generic sg2 type 0
[144873.570444] sd 7:0:0:0: [sdb] 2049 512-byte logical blocks: (1.05 MB/1.00 MiB)
[144873.572853] sd 7:0:0:0: [sdb] Write Protect is off
[144873.572859] sd 7:0:0:0: [sdb] Mode Sense: 03 00 00 00
[144873.579141] sd 7:0:0:0: [sdb] No Caching mode page found
[144873.579146] sd 7:0:0:0: [sdb] Assuming drive cache: write through
[144873.593854]  sdb: sdb1
[144873.594852] sd 7:0:0:0: [sdb] Attached SCSI removable disk

```

As you can see in this log there was 3 plug/unplug of RP2040-Zero to USB and the first time `Write Protect was off` then `Write Protect was on` and after `Write Protect was off` again.

How to have `Write Protect is off` always off?

Thanks.

dhalbert, 2025-04-07:
If a write was in progress when you unplugged or hard-reset the board, the filesystem might have gotten corrupted, and then would come up as read-only. Do a `sync` before that, ideally, or wait a few seconds.

iz8mbw, 2025-04-07:
No read and write was in progress since the usb volume (`/dev/sdb1`) was  never mounted.

tannewt, 2025-04-07:
Linux does read the filesystem before a mount happens. I think it is inspecting the FS for the label for example.

Please try the absolute latest builds too. I've been tweaking this behavior in the repo.