adafruit/circuitpython #10233: board pins initialization?

stanelie, 2025-04-08:
### CircuitPython version and board name

```python
Hello.
Not sure if this is a bug report or a feature request, but I was wondering if there is a way to prevent IO pins from changing state randomly while the device boots?

I use these ESP32-S3-Zero boards to build wireless light sources, I use 4 pins as PWM outputs to drive some LED drivers. While the ESP32 boots circuitpython, before my program runs, the pins go through a cicle of full on - full off, which makes my LEDs go full on while the ESP boots. In my case, in a professional theater context, this is undesired behavior. Also, when I experience low power conditions, when the battery is drained, the ESP32 reboots constantly, which makes the LED blink on each cycle, which is even worse. Is there a way to avoid this (without having to add component to physically enable the outputs only when the device is properly booted)?

Thanks!
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

stanelie, 2025-04-08:
Oh!
That's a good idea, and it won't need to many extra components. I like this!

Thanks!!

dhalbert, 2025-04-08:
Let us know if you are still seeing glitches after that!

stanelie, 2025-04-08:
Very nice! It works perfectly!

Thanks again!!

stanelie, 2025-06-18:
Hello Sir.

I now have the same issue but with a ESP32-C3_mini module this time, and the pull-down resistor has no effect.

Are you aware of something different that could be going on with this chip?

stanelie, 2025-06-18:
Never mind, I was hooked up to the tx/rx pins... Bad choice on my part...