adafruit/circuitpython #10235: Cleveland Art PyPortal Project Exhausts Pystack

FoamyGuy, 2025-04-08:
The project from this learn guide: https://learn.adafruit.com/cleveland-museum-of-art-pyportal-frame

Is unable to run on CircuitPython 9.2.7 on either Pyportal or Pyportal Titano. I did not try earlier versions to find the last working one, but could if that is helpful.

Here is its output include stacktrace. The project code wraps the exception with a message about the SDCard but the rest of the trace shows the root cause is pystack exhausted in the network stack libraries. 

```
code.py output:
retrieving url: https://openaccess-api.clevelandart.org/api/artworks?cc0=1&has_image=1&indent=2&limit=1&skip=8006
Connecting to AP NetworkName
Retrieving data...Reply is OK!
image dim: 771 900
original URL: https://openaccess-cdn.clevelandart.org/1961.202.i/1961.202.i_web.jpg
convert URL: https://io.adafruit.com/api/v2/Foamyguy/integrations/image-formatter?x-aio-key=80b22599ca5d46f28da0705084a14f86&width=158&height=225&output=BMP16&url=https://openaccess-cdn.clevelandart.org/1961.202.i/1961.202.i_web.jpg
Fetching stream from https://io.adafruit.com/api/v2/Foamyguy/integrations/image-formatter?x-aio-key=80b22599ca5d46f28da0705084a14f86&width=158&height=225&output=BMP16&url=https://openaccess-cdn.clevelandart.org/1961.202.i/1961.202.i_web.jpg
Traceback (most recent call last):
  File "adafruit_connection_manager.py", line 61, in connect
  File "adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 114, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 922, in socket_connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 818, in socket_open
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 430, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 337, in _send_command
RuntimeError: pystack exhausted

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/lib/adafruit_pyportal/network.py", line 183, in process_image
  File "adafruit_portalbase/network.py", line 312, in wget
  File "adafruit_requests.py", line 711, in get
  File "adafruit_requests.py", line 639, in request
  File "adafruit_connection_manager.py", line 347, in get_socket
  File "adafruit_connection_manager.py", line 249, in _get_connected_socket
  File "adafruit_connection_manager.py", line 63, in connect
OSError: [Errno 12] ENOMEM: pystack exhausted

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 65, in <module>
  File "/lib/adafruit_pyportal/__init__.py", line 328, in fetch
  File "/lib/adafruit_pyportal/network.py", line 186, in process_image
OSError: 

No writable filesystem found for saving datastream.
                    Insert an SD card or set internal filesystem to be unsafe by
                    setting 'disable_concurrent_write_protection' in the mount options in boot.py

Code done running.
```

Feel free to move this issue to where ever it would be most appropriate, I opened in the core because I tried with latest stable release 9.2.7 and many of the networking libraries installed are frozen into the build.

RetiredWizard, 2025-04-08:
Did you try adding something like `CIRCUITPY_PYSTACK_SIZE=4000` to settings.toml by any chance?

dhalbert, 2025-04-08:
If you put `CIRCUITPY_PYSTACK_SIZE=2048` in `settings.toml`, does it work, and if not `2048`, what size works? There is an issue for this, #10189, but we deferred it to 10.0.0. At the very least we should put a warning in the guide.

FoamyGuy, 2025-04-08:
It does work with `CIRCUITPY_PYSTACK_SIZE=2048`. Thank you.

I'll close this, and for now we can note this in the guide and it's example settings.toml.

PR #10236 (2025-04-09): increase default PYSTACK size from 1536 to 2048
- Fixes #10235.
- Fixes #10189.

Increase default CIRCUITPY_PYSTACK_SIZE from 1536 to 2048, except on SAMD21. This fixes pystack-exhausted errors for several PyPortal projects, and doesn't use up too much RAM.
closed 2025-04-08 by FoamyGuy