adafruit/circuitpython #10237: 10.0.0 Alpha-2 created 2 drives on Pico W

b-blake, 2025-04-08:
### CircuitPython version and board name

```python
See AdaFruit Forum:
10.0.0 Alpha-2 created 2 drives

Adafruit CircuitPython 10.0.0-alpha.2 on 2025-04-04; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
```

### Code/REPL

```python
Any code.py
```

### Behavior

10.0.0 Alpha-2 creates 2 drives on Pico W.  One showing the files.  The other waiting for the user to insert an SD card in the USB drive.  The Pico W does not have an SD card slot.

### Description

Extra SD drive

dhalbert, 2025-04-08:
Originally reported here: https://forums.adafruit.com/viewtopic.php?t=217730.

@tannewt My impression is that this should not be happening on this board?

tannewt, 2025-04-08:
Yup, seems like a bug.

tannewt, 2025-04-08:
Looks like we condition the second drive on CIRCUITPY_SDCARDIO but not the stricter automounting test:

https://github.com/adafruit/circuitpython/blob/63815c5f4522a55a3a06ec706673f952d760d46a/supervisor/shared/usb/usb_msc_flash.c#L31-L38

I don't know why the drive letter would change.

b-blake, 2025-04-08:
@tannewt,

The The drive letter may have been changed because I connected another MCU to the USB cable I was using to see what would happen.   It came up as O:.  When I reconnected the Alpha board to the same cable the drive letters were swapped.  HTH

b-blake, 2025-04-10:
Apparently the AF Feather RP2350 does it also.

tannewt, 2025-05-01:
Thinking about this more, it is intentional because you can attach an SD card breakout to it. The drive without a medium present is intentional in this case.

closed 2025-05-01 by tannewt

b-blake, 2025-05-02:
Is there a possibility to make this feature optional?  It is a PITA for me.

tannewt, 2025-05-05:
Why is an extra drive a pain?

b-blake, 2025-05-06:
@tannewt,

Yes,  I have a bunch of MCUs running and if each uses two drives I will run out of drives.  I will admit I am an unusual case, but here I am. A:, B:, C:, D:, E:, F:, G:, Y:, Z: are used by my system.  That leaves 17 drive letters, H: through X:.  At two drives per MCU that sets the limit to 8 MCUs.   I can't guarantee MCU 9 will get a letter or an empty. This will lead me to not adopt CP 10.x.x.

Some MCUs are running clocks (GPS and NTP), decoration, calibration (time, temperature), environment, entertainment, new AdaFruit hardware, testing, etc.

It is also annoying that the empty drive does not identify what it is affiliated with.

My suggestion would be to put the option to not create the empty drive in the boot.py file.

Bruce

bablokb, 2025-05-06:
You don't need drive letters in Windows to mount something. You can just mount to an existing directory. This is not well known, but Google is your friend.

tannewt, 2025-05-07:
@b-blake If you compile your own without `CIRCUITPY_SDCARDIO` then the second drive won't appear.

b-blake, 2025-05-07:
@tannewt (Scott),

You have wayyyy too much confidence in my abilities.  The last time I compiled code was for the 6502 CPU. back in the 70s and 80s.  My first programming experience was using Fortran in the mid 60s on the IBM 1620. (Not Fortran II, III, IV)  I also learned Assembly on that machine.  I had it compute e to 400 decimal places.  It took hours!

Bruce

P.S. @bablokb,  You too. ;-) Wayyyy too much confidence.

P.S.S.
It is the job of the programmer to make the user's life easy, not the other way around.
Please don't take this personally, but not everybody is as talented/knowledgeable as you guys.  
Most of us just stumble forward as best as we can.

b-blake, 2025-05-08:
Scott,

What are the usage cases for the "thumb Drive" SD card access by the PC? 
Would the PC and the MCU both have simultaneous access to the SD card?

Bruce

bablokb, 2025-05-08:
> P.S. [@bablokb](https://github.com/bablokb), You too. ;-) Wayyyy too much confidence.

Why? There is no programming involved. When you google for "windows more than 26 drives" you will find a lot of instructions on how to do that. Given the fact that you setup so many devices I have a lot of confidence that you can read and follow those instructions. It is about Windows, so it is clicky-clicky and not programming.

tannewt, 2025-05-08:
> Scott,
> 
> What are the usage cases for the "thumb Drive" SD card access by the PC? Would the PC and the MCU both have simultaneous access to the SD card?
> 
> Bruce

The idea is that CP can save to the SD card and then the host can read it. We can also now switch it the other way so the host can load assets onto the SD card for CP to read.

b-blake, 2025-05-08:
Scott,

How about a third option where nobody can write/read it and it goes away?  I know I am being a bit flippant, but...

Will it be switchable real time from within CP.  e.g.  Red LED, write by CP.  Green LED, write by PC.  Red or Green read by both.
  I assume a status can be tested for completion of any write before code.py flips the LED colors. (Two LEDs, Bi/Tri color LED, or NeoPixel)

Default or boot.py or code.py option.
   SD_Card = 0:  No write/read Drive gone.  Default and available from boot.py and code.py.
In boot.py and/or CP options.
   SD_Card = 1:  Write from CP, Read from CP and PC
   SD_Card = 2:  Write from PC, Read from CP and PC
   SD_Card = 3:  Read only from CP and PC.  Don't leave it open for write if you don't need to write.

You are starting to win me over, but I still would like SD_Card = 0 as an option.

Bruce

tannewt, 2025-05-09:
I don't want separate controls for the SD card drive. However, I would like it to be omitted if you do [`storage.disable_usb_drive()`](https://docs.circuitpython.org/en/latest/shared-bindings/storage/index.html#storage.disable_usb_drive) If it doesn't, then let me know. That'll clean up all the drives you have from the multitude of CP devices you have connected.

When I do that, I usually have a button I can press on startup to make it appear. Safe mode can be used for that too.

b-blake, 2025-05-09:
Scott,

storage.disable_usb_drive would disable CIRCUITPY and the thumb drive.  I guess I stay at 9.2.7.
You are looking for easy for you.  I am looking for easy for me.  We can't have both, and you are driving.