adafruit/circuitpython #10247: Unable to get into .UF2 boot mode from code

ThomasAtBBTF, 2025-04-13:
With (much) older versions of CircuitPython I was able to get into .UF2 boot mode from inside a running CircuitPython app.
The current instructions / documentation in circuitpython.org for version 9 suggest to code like so:
 import microcontroller 
 microcontroller.on_next_reset(microcontroller.RunMode.UF2)
 microcontroller.on_next_reset()
However, this currently does not work for me for any ESP32S3 boards I have access to.

Any ideas ?

Thomas

Neradoc, 2025-04-13:
I would ask: _what does "does not work" mean ?_ But you have the last line wrong, so if that's the code you are actually using (and not a typo here), you would have gotten a `TypeError: function takes 1 positional arguments but 0 were given` exception.

This is what you want:
```py
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.UF2)
microcontroller.reset()
```
It works for me (tested on a TFT feather and a QTPY).
If you code was correct and you still get the issue, what board are you using exactly ?

ThomasAtBBTF, 2025-04-13:
I tried this with a:
- adafruit_feather_esp32s3_reverse_tft
- firebeetle2_esp32s3
- yd_esp32_s3_n16r8
I tried with all of them under 9.2.7
With the firebeetle2_esp32s3 I also tried 9.0.5 to see for me if it was working in an earlier version but it did't

What does not work mean: 
- all three board are not showing a USB drive after the reset I "see" the debug com port.
- I do not see a Typerror Message!

Here what I get in PUTTY:
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.2.7 on 2025-04-01; DFRobot FireBeetle 2 ESP32-S3 with E
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.UF2)
>>> microcontroller.reset()

Immediately after reset() the serial connection is lost and Putty brings up a error message!

Also: I am asking ths in part also because with all three boards I can not get into the .UF2 "drive by pressing the BOOT / RESET combination. And "Yes" I got this working on other boards.

ThomasAtBBTF, 2025-04-13:
Additional info:
When I call:
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
microcontroller.reset()

I get a "new" COM-Port and a "bad" ESP32S3 device in the Windows "device manager"

Neradoc, 2025-04-13:
> Also: I am asking this in part also, because with all three boards I can not get into the .UF2 "drive by pressing the BOOT / RESET combination.

Is the UF2 bootloader even installed ? How did you install Circuitpython if you can't get into the UF2 bootloader ? The installer on the board page ? (you want the full/uf2 install) Or did it work and no longer does ?

ThomasAtBBTF, 2025-04-13:
I will try again with CP 9.2.7 I falsely stated above that I tested with the Reverse TFT Feather with 9.2.7 but when I checked it was version 9.2.4.

I updated the bootloader from https://circuitpython.org/board/adafruit_feather_esp32s3_reverse_tft/ using the open installer.
After a manual reset the drive FTHRS3BOOT appeared.

Also when typing now:
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.UF2)
microcontroller.reset()

The FTHRS3BOOT appears.

Thank you snkYmkrct for mentioning the bootloader !

ThomasAtBBTF, 2025-04-13:
I have no idea how I got "rid of" the bootloader on the adafruit_feather_esp32s3_reverse_tft

On the firebeetle2 there is no bootloader option on https://circuitpython.org/board/firebeetle2_esp32s3/
So I uploaded firmware.bin at pos 0 with the help of the website: https://adafruit.github.io/Adafruit_WebSerial_ESPTool/
 Can anyone give me idea how / where I can find the bootloader for the firebeetle2 and with what parameters I need to upload the UF2 Bootloader into the board.

snkYmkrct, 2025-04-13:
You can download the UF2 bootloader from [the tinyuf2 repo](https://github.com/adafruit/tinyuf2/releases/tag/0.30.0)
Not sure why it's not linked on the download page.

ThomasAtBBTF, 2025-04-13:
After using combined.bin from tinyuf2-firebeetle2_esp32s3-0.30.0.zip and rebooted, the drive FIRE2BOOT appeared.
Then I copied the CP 9.2.7 .UF2 file into the FIRE2BOOT drive.
After a reset the CIRCUITPY drive appeared and I can confirm that the sequence:

import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.UF2)
microcontroller.reset()

Brings me again into the FIRE2BOOT drive!

Great!

One small thing remains and this may be specific to the firebeetle2:

When I press the BOOT RESET combination, the board does not come up with the FIRE2BOOT drive but the COM-Port comes up with the USB / JTAG serial debug unit (COM46 in my case)

snkYmkrct, 2025-04-13:
Holding BOOT and pressing RESET puts the board in the ROM bootloader. 

The UF2 bootloader should start by double pressing RESET -- you may need to try a few times to get the timing right -- see more details [here](https://learn.adafruit.com/esp32-s3-reverse-tft-feather/install-circuitpython#circuitpython-quickstart-3096177)