adafruit/circuitpython #10248: RP2350 Never Runs Boot Stage 2

eightycc, 2025-04-14:
At startup, RP2350-based boards do not run boot stage 2. This is due to a change in the way `_stage2_boot()` is invoked. For the RP2040, boot stage 2 was invoked directly by boot ROM. For the RP2350, stage 2 is invoked by `crt0.S` under the control of `PICO_EMBED_XIP_SETUP`. Since CP does not set `PICO_EMBED_XIP_SETUP`, it defaults to 0, causing `_stage2_boot()` to not be copied onto the stack and executed. See https://github.com/raspberrypi/pico-sdk/issues/1903.

RP2350 boot ROM tests flash, selecting XIP parameters automatically, including the clock divider and one of four XIP modes (two serial modes, a dual mode, and a quad mode). This obviates the need for boot stage 2 for normal clocking cases. Since CP has been running successfully without boot stage 2, it's clear that it is not needed at this time.

However, this could be an issue for overclocking and flash writes as discussed in the referenced Pico SDK issue. Boot ROM exits from flash writes with flash timing and mode set to its automatically determined values overriding any changes that may have been made for overclocking.

I'm filing this issue so that this detail doesn't get lost.

dhalbert, 2026-11-03:
Investigate whether this is important or not.

dhalbert, 2026-11-14:
Hi @mikeysklar I started on this anyway, and it turned into more, with a lot of decisions. Will have a PR in the near future.

closed 2026-11-16 by dhalbert