adafruit/circuitpython #10256: Error allocating DMA buffers for I2SOut playback

relic-se, 2025-04-15:
### CircuitPython version and board name

```python
# Tested without error
Adafruit CircuitPython 9.2.7 on 2025-04-04; Waveshare RP2040-PiZero with rp2040
Adafruit CircuitPython 10.0.0-alpha.2 on 2025-04-04; Waveshare RP2040-PiZero with rp2040

# Tested with error
Adafruit CircuitPython 10.0.0-alpha.2-16-g8f7496c069 on 2025-04-15; Waveshare RP2040-PiZero with rp2040
```

### Code/REPL

```python
# waveshare rp2040-pizero with waveshare audio hat (requires i2c pull-ups)

import time
import board
import digitalio
import audiobusio
import synthio
from adafruit_wm8960 import WM8960

# Configure codec
codec = WM8960(board.I2C(), 11025, 16)
codec.volume = 1.0
codec.headphone = 0.5
codec.speaker = 0.0

# Configure I2S Output
audio = audiobusio.I2SOut(board.GP18, board.GP19, board.GP21) # ADC is GP20

# Setup synthio
synth = synthio.Synthesizer(sample_rate=codec.sample_rate, channel_count=2)
audio.play(synth)

note = synthio.Note(220, panning=-1.0)

while True:
    print("note on")
    synth.press(note)  # midi note 65 = F4
    time.sleep(0.5)
    synth.release(note)  # release the note we pressed
    print("note off")
    time.sleep(0.5)
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
RuntimeError: Unable to allocate buffers for signed conversion
```
No audio output occurs.

### Description

- The source of the error message is https://github.com/adafruit/circuitpython/blob/8f7496c069956c7fe7ef59c55e35310093703343/ports/raspberrypi/common-hal/audiobusio/I2SOut.c#L276
- The error flag likely occurs at https://github.com/adafruit/circuitpython/blob/8f7496c069956c7fe7ef59c55e35310093703343/ports/raspberrypi/audio_dma.c#L234
- The most recent change to the audio DMA is from https://github.com/adafruit/circuitpython/pull/10240. I haven't verified if those changes are definitively the cause of this error.

### Additional information

I don't completely understand what's going on in `port_realloc`, but the primary difference I can discern against `m_realloc` is that is uses `tlsf_realloc` instead of `realloc`.

Any thoughts on this? @tannewt

relic-se, 2025-04-15:
@tannewt I can confirm that reverting the changes made to `ports/raspberrypi/audio_dma.c` (using `m_realloc` and `m_free` instead of `port_realloc` and `port_free`) fixes the error. If moving over to the `port_...` implementation is better for future-proofing, there is likely an underlying issue that needs to be fixed rather than reverting those changes.

tannewt, 2025-04-16:
I suspect this is due to the Python VM heap taking up most of the outer heap (allocated via port_malloc using tlsf). It might be simplest to use `m_malloc` on RP2040 and `port_malloc` on RP2350. The goal is to put the buffers into internal RAM. DMAing from PSRAM gums up DMA.

relic-se, 2025-04-17:
I'm testing now on a Pimoroni Pico Plus 2 with 8MB of PSRAM. Is this basically what we're thinking of? I can get a PR up for this, but I want to do a bit more testing.

```c
#ifdef PICO_RP2350
dma->buffer[0] = (uint8_t *)port_realloc(dma->buffer[0], max_buffer_length, true);
#else
dma->buffer[0] = (uint8_t *)m_realloc(dma->buffer[0],
    #if MICROPY_MALLOC_USES_ALLOCATED_SIZE
    dma->buffer_length[0], // Old size
    #endif
    max_buffer_length);
#endif
dma->buffer_length[0] = max_buffer_length;

if (dma->buffer[0] == NULL) {
    return AUDIO_DMA_MEMORY_ERROR;
}
```

```c
#ifdef PICO_RP2350
port_free(dma->buffer[0]);
#else
#if MICROPY_MALLOC_USES_ALLOCATED_SIZE
m_free(dma->buffer[0], dma->buffer_length[0]);
#else
m_free(dma->buffer[0]);
#endif
#endif

dma->buffer[0] = NULL;
dma->buffer_length[0] = 0;
```

tannewt, 2025-04-21:
Yup! That's what I'm thinking.

closed 2025-04-22 by tannewt